marcschmiedchen/home-assistant-wolf_ism8 #1: Unknown data points

nanosonde, 2021-11-16:
Hi!

I only have a Wolf CWL Excellent 300 with an ISM8 and can see the following in the logs:
```
 Logger: wolf_ism8.ism8
Source: /usr/local/lib/python3.9/site-packages/wolf_ism8/ism8.py:394
First occurred: 13:31:37 (31 occurrences)
Last logged: 13:31:39

    unknown datapoint: 354, data:0
    unknown datapoint: 355, data:0
    unknown datapoint: 356, data:0
    unknown datapoint: 357, data:255
    unknown datapoint: 358, data:0

```

Any idea what this might be?

Do you have any plan to support these?
```
 Logger: wolf_ism8.ism8
Source: /usr/local/lib/python3.9/site-packages/wolf_ism8/ism8.py:494
First occurred: 13:31:33 (10 occurrences)
Last logged: 13:31:36

    datatype not implemented: DPT_Date
    datatype not implemented: DPT_TimeOfDay
    datatype not implemented: DPT_FlowRate_m3/h
```

At least the flow rate....

BTW: Where can I find the repo for the wolf_ism8 python module?

marcschmiedchen, 2021-11-17:
Hi,

The "unknown datapoint" log-entry means, that ISM8 is receiving a datapoint-number (354,355, etc) which was was not documented at the time I wrote the library. See pdf in attachment. There is probably a newer version now.

The unknown datatype means, that I did not implement those datatypes (Date, TimeofDay,Flowrate) at the time, because I didnt need them for my Setup back then. I might correct both flaws in a couple of days, so you can read all the values for your CWL

But I'm not planning to implent the send-feature at the time...

The library is hosted on Pypi. https://pypi.org/project/wolf-ism8/

regards
MS

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

nanosonde, 2021-11-17:
Ok, thanks for the reply.

I have already studied various versions of the Wolf PDF.
I wonder if those datapoint IDs really exist or if this is a parsing/decoding error in wolf_ism8.

I can understand that you do not want to implement writing to datapoints.
However, the request for all datapoint updates is just a fixed telegram that does not need any encoding or the like.
We would just need to send this fixed byte sequence after every successfull connection event and that's it. 

Concerning the source code:
I have already found wolf-ism8 on Pypi. However, I did not find the GIT repo where to place pull requests.

marcschmiedchen, 2021-11-18:
hi,
I changed the Library-Repo to public on Github ("wolf_ism8").
As for CI/CD Integration, I will look into it.

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

nanosonde, 2021-11-18:
Already got it working: https://github.com/nanosonde/python-wolfism8/tree/main/.github/workflows

marcschmiedchen, 2021-11-20:
As for the unknown datapoints : Unlikely that there is a parser error in the library....
Datapoint numbers are sent as a 2-byte value. Decoding is straightforward, and all Datapoints up to 210 are implemented and work flawlessly. 354 etc as DP-ID seems reasonable, but is not documented. I can implement them as "unknown#xxDP_ID", if you like.... at least then they are not discarded?

marcschmiedchen, 2021-12-08:
Implemented missing datapoints as "undocumented354....358".

closed 2021-12-08 by marcschmiedchen