marcschmiedchen/home-assistant-wolf_ism8 #3: Implement DPT_FlowRate for CWL

nanosonde, 2021-11-16:
DPT_FlowRate decoder in PHP:

```php
function convfrom_DPT_FlowRate($val)
{
	$value = ($val[0]<<24)|($val[1]<<16)|($val[2]<<8)|($val[3]<<0);
	$flowrate = $value * 0.0001;
	return ((string)$flowrate);
}
```

marcschmiedchen, 2021-11-20:
Tried to implement it.  Issue moved to wolf-ism8 library. Please check for functionality and report back.

closed 2021-11-20 by marcschmiedchen

nanosonde, 2021-11-20:
Hi!
Stupid question, but how do I just update the wolf_ism8 module. The custom component did not change or require a newer version of it.
I am using HASS OS.

nanosonde, 2021-11-22:
@marcschmiedchen 

The flowrate is now shown. The value is correct, but the unit of measurement is wrong.
Home Assistant shows "ccm/h". This is wrong. It is supposed to be  "m3/h".