marcschmiedchen/home-assistant-wolf_ism8 #5: Installation from Pund setup

stephangeiger1, 2021-12-02:
Hallo marcschmiedchen,

ich versuche gerade die WolfISM8 Software in HA zu intergrieren. Leider scheitere ich schon beim Versuch die Dateien über Python (vonPyPi) zu installieren. Mein HA läuft auf einer VM mit dem OS Betriebssystem.

Vielen Dank vorab
Stephan

Hello marcschmiedchen,

I am trying to integrate the WolfISM8 software into HA. Unfortunately I fail already when trying to install the files via Python (fromPyPi). My HA is running on a VM with the OS operating system.

Thanks in front
Stephan

marcschmiedchen, 2021-12-03:
öh. 
Das sind ein bisschen wenig Infos.  Was passiert denn beim Befehl:
"pip install wolf_ism8"

stephangeiger1, 2021-12-10:
Hallo,
wenn ich den "pip" befehl über die Shell eintippe, bekomme ich nachfolgendes: 
bash: pip: command not found

lukastripp, 2021-12-16:
Hallo Marc,
ich finde den Adapter echt super den du geschrieben hast. Leider scheitere ich auch an der Installation. Wie Stephan habe ich auch kein "pip" auf der cli verfügbar. Auch ich habe HA OS in einer VM. Gleicher Fehler mit "PIP". Könntest du das ganze genauer dokumentieren? Fände es schade wenn deine Arbeit nicht wertgeschätzt würde. Suche schon lange nach einem Adapter der ism8 unterstützt.

Auch "source <put here your homeassistant_directory>/bin/activate" das vielleicht etwas genauer. Speziell für Leute die sich einfach nicht gut genug auskennen, um das einfacher zu implementieren.

marcschmiedchen, 2021-12-16:
hi,
ich hab eure probleme mal durch-ge-googelt und die zusammenfassung ist: HA-OS in einer VM hat einfach kein PIP installiert und es scheint generell eher schwieriger/komplizierter, solche custom-components nachzuinstallieren. Ich konnte allerdings nur mit google auf die Schnelle auch nicht rausfinden, was genau das problem ist, bzw. woran es liegt; ich weiß nichtmal, wie sich HA OS bedient.

Ich betreibe "Home Assistant Core", das ist die selbst installierte Version auf einem Raspi, ohne VM. Da kommt man gut an alles ran und kann auch Pakete installieren, etc. Auch das mit dem /source/bin/activate ist sehr spezifisch für HA Core, generell schirmt man damit seine Python Installation vom Rest des Systems ab. Dazu gibts wirklich viele Erklärungen, googelt mal "venv". Das Problem ist, ich weiß nicht, wie das unter HA OS ist.

Ich würde ja gerne helfen, aber dazu müsste ich selbst mal komplett eine neue Maschine aufsetzen, HA OS installieren und dann gucken, was sache ist. Dazu hab ich im moment keine Zeit, aber WENN ich es mal mache, dann schreibe ich es hier. Vielleicht koennt ihr eure Fragen auch in einem Forum beantwortet bekommen. Generell wäre die Frage "how to install python modules in HA OS".....

Bin ziemlich erstaunt, dass man dazu nicht direkt eine einfache antwort bekommt, denn es gibt sehr viele insel-lösungen, die auf GITHUB gehostet werden. Keine Ahnung, wie die installiert werden, die müssen ja alle das gleiche Problem unter HA OS haben.

Ein Tipp zuletzt: überspringt doch einfach mal alle schritte, die nicht klappen. Vielleicht ist HA OS ja so gut, dass es selbst weiß, was stattdessen zu tun ist (wäre möglich). Irgendwie müsst ihr aber definitiv meine Dateien in dieses "custom-components" Verzeichnis packen und auch die Datei "configuration.yaml" bearbeiten,....

nanosonde, 2021-12-16:
Hi!
Ich nutze das HA OS.
Man muss da gar nichts von Hand installieren.
Das macht HA alles selbst.
Im Python Code dieser Custom Integration steht ja explizit für HA drin, welches zusätzliche Python Modul zusätzlich installiert werden muss.

Habe einfach in HACS ein Custom Repo hinzugefügt. Nämlich das der Wolf HA Integration. Dann nur installieren und config anlegen. Fertig.

nanosonde, 2021-12-16:
@marcschmiedchen 
BTW:
https://developers.home-assistant.io/docs/creating_component_code_review

Requirements have been added to manifest.json. The REQUIREMENTS constant is deprecated.
Requirement version should be pinned: "requirements": ['phue==0.8.1']

marcschmiedchen, 2021-12-16:
Aha, super tip, da lohnt sich die Community. @stephangeiger1 und @lukastripp: HACS ist ein integrationsdienst fuer nicht-offizielle Repos. Ich weiss nicht, ob der schon in HA OS integriert ist, oder selbst wieder erstmal installiert werden muss (@nanosonde, vielleicht kannst du den beiden dazu nochwas sagen). Ich stelle auf jeden Fall alle diesbezueglichen Installationstips , die hier ausgetauscht werden, auf die Haupt-Page / ins README.  Auch wenn einer von den Neu-Installateuren eine Anleitung schreiben will: alles was ihr mir schickt, stelle ich fuer andere mit aehnlichen Anlaufschwierigkeiten auf die Github Page

das mit den requirements aendere ich.

nanosonde, 2021-12-16:
HACS ist nicht standardmäßig in HA OS enthalten.
Das lässt sich aber ganz einfach anhand der HACS Anleitung erledigen.

Alternativ kann man diese Integration aber auch von Hand in den Ordner „custom_components“ im Config-Ordner installieren. Einfach einen Unterordner erstellen und da alles reinkopieren. Dann HA OS einfach neu starten.
Dann benötigt man HACS gar nicht. Hat dann aber auch keine komfortablen Updates der Custom Integrations.

Ich empfehle auf jeden Fall die Installation des Samba Addons, um komfortabel von Windows aus auf die wichtigen Ordner von HA OS zugreifen zu können.

stephangeiger1, 2021-12-16:
Hallo Zusammen,
ich möchte mich für den regen Support bedanken. 
Danke @marcschmiedchen für die aufwändige Recherche und vielen Dank an @nanosonde für die Anleitung zur Installation über HACS. 
Ich habe die ISM 8 Software wie von nanosonde beschreiben via HACS installiert und die Konfiguration im HA vorgenommen.
Die Integration ist nun installiert im im Lovelace Board habe ich mir bereits einige Entitäten anzeigen lassen.

Leider bekomme ich bisher keine Werte vom ISM8 zurück.

Die Zielserver IP habe ich auf die IP Adresse gesetzt, die mir meine HomeAssistant Installation zurück gibt.
Wie nachfolgend
IPv4 adresses for enp0s3: 192.168.100.80/24, die 
Zielserverport: 8123

Die /24 habe ich bei der Zielserver Adresse schon weggelassen und auch mit hinzugefügt. Beide mal bekomme ich keine Werte zurück.

Vielen Dank vorab für die Antworten

@lukastripp hast du auch schon was am laufen?

Gruß Stephan

nanosonde, 2021-12-16:
Das ISM8 übernimmt die IP Einstellungen erst nach einem Neustart.
Ich mache es dazu kurz stromlos.

nanosonde, 2021-12-16:
Bei mir ist übrigens als Port 12004 eingestellt. Das ist default, glaube ich.

stephangeiger1, 2021-12-16:
Hallo nanosonde,
danke, das ISM8 habe ich stromlos gemacht.
Ich versuche nochmal Port 12004

stephangeiger1, 2021-12-16:
@nanosonde ,
es läuft :)
Vielen Dank für die Unterstützung. Ich habe bereits ein paar Werte erhalten.  Mal schauen, wie es morgen aussieht.

lukastripp, 2021-12-17:
Bei mir geht es auch per Custom Repo in HACS. Warum bin ich nicht früher drauf gekommen. Läuft 1a das Plugin. Nochmal vielen Dank an @nanosonde und @marcschmiedchen für deine Arbeit!!!

lukastripp, 2022-01-08:
Ich habe jetzt ein neues Problem. Nach dem ich meiner HA VM einen WLAN USB verpasst habe um per MQTT meinen Pelletofen mit dem eigenen WLAN Netz auszulesen, kommen keine Werte mehr an. Problem ist das der Wolf Adapter auf dem WLAN Adapter horcht intern. Laut Doku kann ich ja bei mehreren IPs meiner HA VM das in der Konfig mit geben.
Meine Konfig:
# wolf CGB-2 Heizung  
wolf:
    devices:
      - HG1
      - BM1
    host:
      - 10.200.2.251
    port:
      - 12004
      - 
Nur dann bekomme ich den Fehler nach Konfig prüfen:

Invalid config for [wolf]: expected int for dictionary value @ data['wolf']['port']. Got [12004]
value should be a string for dictionary value @ data['wolf']['host']. Got ['10.200.2.251']. (See /config/configuration.yaml, line 73). 

Jmd  @stephangeiger1 @nanosonde  auch schon das Problem gehabt?
@marcschmiedchen  ist das nur ein Doku Thema bei dir? Oder muss ich noch den Werte Typ per cast ändern? Ist natürlich jetzt ein spezifisches HA Thema.

marcschmiedchen, 2022-01-09:
> Jmd ***@***.***(https://github.com/stephangeiger1) ***@***.***(https://github.com/nanosonde) auch schon das Problem gehabt?

Mein fehler in der Doku./ falsches config format.

Es muss heißen:
wolf:
devices:
- HG1
- BM1
host: 10.200.2.251
port: 12004

habs auf github korrigiert.

Gruss
MS

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2022-01-10 by stephangeiger1