marcschmiedchen/home-assistant-wolf_ism8 #6: Unbekannte Daten für Heizgerät HG1- LWP BWL1S07

stephangeiger1, 2022-01-10:
Hallo,

ich bekomme vom Heizgerät HG1den Wert: "Unbekannt" für fast alle Parameter vom ISM8 Modul zurückgeliefert. 
Ich habe das ISM8i an einer Luft-Wärmepumpe BWL1S07 hängen.

- hg1_brennerleistung
- hg1_anlagendruck
- hg1_status_brenner
- hg1_sammlertemperatur
- hg1_warmwassertemperatur
- hg1_kesseltemperatur

Gruß Stephan

stephangeiger1, 2022-01-13:
Hallo,

nach Durchsicht des Codes (bin kein Programmierer) habe ich die Lösung gefunden.
Es werden alle Werte vom ISM8 abgefragt.
Die Luftwärmepumpe ist bereits implementiert (Datenpunkt 176-191). 
Hierzu muss in der config -BWL hinzugefügt werden.

  wolf:
    devices:
      - HG1
      - BWL

closed 2022-01-13 by stephangeiger1

marcschmiedchen, 2022-01-16:
Sorry für die späte Antwort! 
Wie Du schon selbst rausgefunden hast, ist die Konfigurationseinstellung für die Luftwärmepumpe "BWL". Das hatte ich derzeit in der Anleitung vergessen, bzw. weggelassen weil ich gar nicht wusste, was sich hinter "BWL" verbirgt.

Ist aber keine Issue der Bibliothek, sondern der Home-Assistant Config. Habs in der Anleitung ergänzt, danke.