marcschmiedchen/home-assistant-wolf_ism8 #7: Working with ism8i?

j3mu5, 2022-03-03:
Does this solution also work for the ism8i? I have tried it with one and get the following error message:

2022-03-03 21:15:22 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:0
2022-03-03 21:15:22 ERROR (MainThread) [wolf_ism8.ism8] datatype unknown, using INT: DPT_unknown

nanosonde, 2022-03-03:
Basically there is only one ISM8 PCB.
The letters i and e just describe if it is just the bare PCB for mounting inside a Wolf heating device (letter i).
The other variant is just the same PCB, but an additional external housing and some DCDC converter circuity to be able to connect it to a normal 5V USB power supply.
I am also using the ISM8i with my own power supply.

marcschmiedchen, 2022-04-01:
As nanosonde correctly replied, the library works on all ISM8. I myself own an ISM8i. The error message was already mentioned in another issue here  on github-- the documentation of the ISM8, which is provided by Wolf, does not contain certain datapoints. Therefore this error means "Hello, your ISM8 is sending datapoint 363, value 0, but I do not know what that represents". You can safely ignore that error. [I might contact WOLF GmbH if they have any better documentation though]
But all other info should be populated already? If you're at this point, then the connection is established, and home assistant has successfully integrated the custom library? All the sensors should appear shortly after startup !

closed 2022-04-01 by marcschmiedchen