marcschmiedchen/home-assistant-wolf_ism8 #8: Connection from HomeAssistant lost after serveral days

stephangeiger1, 2022-04-06:
Hello,

the connection between HomeAssistant is always lost after a few days. Turning off the power and restarting the ISM8 solves the problem, until after a few days the connection is lost again.

The ISM8 is assigned a static IP.

Regards Stephan

marcschmiedchen, 2022-04-08:
Hi,
this might be a ISM8-hardware or general network issue, nothing specifically related to the library..... I have been using an ISM8(i) for several years now and connection has never dropped, not once....
Troubleshooting tip: 
When the connection is lost once again, try to restart Home Assistant (but not the ISM8). The library will try to re-connect. If that works, the connection loss is/was only temporary and I might be able to fix it with an automatic re-connection attempt inside the library. If a fresh start of home assistant is NOT sufficient, the the problem is with your ISM8 hardware, which I obviously cannot fix.

closed 2022-05-19 by stephangeiger1

stephangeiger1, 2022-10-11:
Hello,

the problem was a switch between the ISM8 and the router.
I connected the ISM8 directly on the LAN Port of the router and the problem was fixed.

Regards Stephan

stephangeiger1, 2024-10-31:
I will try another ISM8 hardware und make an report