marcschmiedchen/home-assistant-wolf_ism8 #10: Nur Lesen, kein Schreiben

chino-lu, 2022-09-12:
**Bug description**
Es ist nur möglich Daten zu lesen. Jedoch nicht zu schreiben. Beispiel: 1x Warmwasser wird als Bool angezeigt, anstatt, wie in der Beschreibung als DPT_Switch.

**Your environment**
 Home Assistant 2022.9.2 
 Supervisor 2022.08.6 
 Operating System 8.5 
 Frontend 20220907.0 - latest 

**Expected behaviour**
Ich kann die 1x Warmwasser per HomeAssistant über Button aktivieren.

**What went wrong?**
1x Warmwasser (und andere) sind als Boolean definiert.

stephangeiger1, 2022-10-11:
Hallo marcschmiedchen,
ich nutze die HomeAssistant Integration nun auch schon einige Zeit und habe durch diese
schon einige Verbesserungen an meiner Heizungregelung vornehmen können. 
Durch die Log's ist es mir auch sehr schnell aufgefallen als Solarkollektor vorzeitig ausfiel.

Ich würde es super finden, auch Werte schreiben zu können und würde mich sehr über eine Implementierung freuen.
Gerade in der aktuellen Energiekrise, liese sich somit noch mehr aus der Heizung herausholen und Kosten einsparen.

Gruß Stephan

tomben22, 2022-10-12:
hallo @marcschmiedchen,
ich habe mir nun auch ein KNX Schnittstellen Set mit dem ISM8i zugelegt und frage mich ob ich darüber überhaupt schreiben kann. Hast du das schon mal erfolgreich getestet? 

Gruß & Danke Tom

stephangeiger1, 2022-10-12:
Hallo Tom,

nachfolgend ein Auszug aus dem Technischen Handbuch des ISM8.
Es ist ein Schreiben über ISM8 möglich.

Gruß Stephan

Mit dem Schnittstellenmodul ISM8i kann der Nutzer Datenpunkte von 
Wolf-System-Komponenten selbstständig per Ethernet verarbeiten. 
Dazu muss softwaretechnisch zunächst eine Verbindung zum Schnittstellenmodul 
aufgebaut werden. IP-Adressen für ISM8i und den Kommunikationspartner können 
über	eine	Weboberfläche	vergeben	werden.	Anschließend	sind	die	Daten	über	das	
spezielle „TCP/IP-Protokoll mit integriertem ObjectServer-Telegramm“ auszuwerten. 
Das ISM8i sendet eigenständig „Request“-Telegramme mit den aktuellen Daten-
punktinhalten über Ethernet. Diese müssen durch „Response“-Telegramme vom Kom-
munikationspartner bestätigt werden. Für die Aktualisierung von Datenpunktinhalten 
sendet der Nutzer ein „Indication“-Telegramm an das ISM8i. 

Es gilt: Variablen, die als „Output“ gekennzeichnet sind (siehe 8.3), sind nur zum 
Lesen gedacht und können vom Nutzer nicht verändert werden. **Variablen, die als 
„Output / Input“ angegeben sind, können dagegen über das entsprechende Telegramm 
an das ISM8i gesendet und als Sollwerte vorgegeben werden.**

chino-lu, 2022-10-12:
hab mir nachträglich eine Weinzierl Baos 771 geholt... damit kann man auch schreiben (die parameter die es laut Handbuch zulassen)... leider ging bei mir die verbindung zu Home Assistant ab und an flöten... lief also nicht ganz stabil. Mit der Weinziel BAOS keine probleme bisher.

lukastripp, 2022-11-26:
> hab mir nachträglich eine Weinzierl Baos 771 geholt... damit kann man auch schreiben (die parameter die es laut Handbuch zulassen)... leider ging bei mir die verbindung zu Home Assistant ab und an flöten... lief also nicht ganz stabil. Mit der Weinziel BAOS keine probleme bisher.

Könntest du das mal genauer ausführen? Ich habe nichts mit KNX IP im Einsatz zu hause. Würde mir aber, wenn das Ding "Werte schreiben" ermöglicht, eins zu legen. Ist es dann auch in HA integriert sauber? Für die Community wäre es toll auch die Programmlogik die zum Schreiben dort genutzt wird zu bekommen.

stephangeiger1, 2022-11-30:
Hallo,
im Fork von **Maxibick**, könnnen Werte an das ISM8 gesendet werden. Leider **fehlen** hier ein paar Datenpunkte die nicht als Sensoren ausgegeben werden, die in der Integration von marcschmiedchen aber schon implementiert sind.
z.B. bei **Wärmepumpe BWL**
- 177 - Betriebsart
- 178 - Heizleistung
- 179 - Kühlleistung
- 186 - Status Zubringer Heizkreispumpe
- 190 - Anlagendruck
- 191 - Leistungsaufnahme

Besonders der letzte Datenpunkt (191 - Leistungsaufnahme) liefert sehr hilfreiche Werte.

chino-lu, 2022-12-02:
> Könntest du das mal genauer ausführen? Ich habe nichts mit KNX IP im Einsatz zu hause. Würde mir aber, wenn das Ding "Werte schreiben" ermöglicht, eins zu legen. Ist es dann auch in HA integriert [...]

Der rest unseres SmartHomes basiert auf KNX. Falls das nicht gegeben ist, weiss ich nicht ob es sich Lohnt, da man ein KNX IP Interface + Software (Home version ist glaub ich 600€, ausser man findet jemand der einem das ganze Programmiert, aber auch da kommt man schnell an die 1000€)

Mit der KNX integration kann man halt definieren was nur dargestellt werden (lesen) soll oder eben auch eine Übertragung an das KNX System (schreiben).

Dark-Tower-Coder, 2023-01-31:
Hey.
@marcschmiedchen 
Wäre cool wenn du für zwei oder drei Datenpunkte das Schreiben implementieren könntest.
Ich würde dann anhand deiner Implementation weitere Datenpunkte mir Schreibfunktionen erweitern.

@lukastripp 
Wäre jetzt nur eine temporäre Lösung:
Im ioBroker gibt es auch einen Adapter für die ism8i Schittstelle und da ist das Schreiben überwiegen implementiert.
ioBroker kann man auch als Add-On in HA installieren.
Wäre aber keine schöne Lösung und ich bevorzuge die vollständige Implementation in HA.

marcschmiedchen, 2023-02-04:
Hallo zusammen, danke für das allgemeine Interesse; ich hab Github leider sehr lange schleifen lassen. 
Ich wollte/werde mich mal mit Maxibick in Verbindung setzen und ihn bitten ein pull request zu machen, damit ich seinen Code bei mir integrieren kann, sowohl in der Wolf-Library als auch am Homeassitant_Teil. Er hat die ganze Integration komplett modernisiert, da ich auf einem 2020 HA entwickelt hatte und es da unter der Haube viele Änderungen gab. Also ich muss mir eine Woche oder so Zeit nehmen, mich mit GIT anfreunden und dann kriege ich es hoffentlich hin. Ich berichte dann !

Dark-Tower-Coder, 2023-02-04:
@marcschmiedchen 
Top!
Finde ich klasse von Dir und auch von @maxibick. :-)
Wenn ich in irgendeiner Art unterstützen kann, lass es mich wissen.

maxibick, 2023-02-04:
Meld dich gerne @marcschmiedchen, freu mich drauf!

Meine Erweiterung ist leider noch nicht komplett. Bin bei der Implementierung von Hölzchen auf Stöckchen gekommen und habe irgendwann festgestellt, dass es ein Software-Update des ism8 gab. Um das zu bekommen musste ich das ganze Hardware-Modul über meinen Klempner bei Wolf einreichen.. Es sollte in den kommenden Wochen wieder kommen, hat ewig da gelegen. :( 

Wie @stephangeiger1 bereits geschrieben hat fehlen noch ein paar Datenpunkte, weil ich mit den verschiedenen Sensortypen gespielt habe... und weil ich keine Wärmepumpe habe. :P

marcschmiedchen, 2023-02-20:
So.
Ich habe in der letzten Woche(n) die grundlegende wolf_ism8 Library von Grund auf neu geschrieben (Hab dafür vor allem bei den Float-Codierungen bei @maxibick abgekupfert -- danke für die Erst-Implementation!) Sie wurde etwas umfangreich, sodass ich das Package auf 3 Dateien aufgeteilt habe, sowie eine Testimplementation fürs Schreiben der Typen Float, Scaling und Bool. Diese drei Typen habe ich auch erfolgreich testen können, und sie decken eigentlich alle beschreibbaren Datenpunkte im ISM8 ab, bis auf Datums.- und Zeitangaben der CWL Lüftungssteuerung; da ich keine besitze hätte ich die Library damit nicht testen können, also hab ich es erstmal gelassen. Wenn jemand unbedingt Datums und/oder Zeiten an die ISM8 schicken will, möge er hier noch ein Issue aufmachen. 
Die neue Lib liegt jetzt auf Pypi in der Version 2.0.

Nächster Schritt wäre die Integration in HA, da teste ich jetzt erstmal die Version von @maxibick und setz mich mit ihm in Verbindung. Das testen dauert aber ein bisschen, ich muss Docker aufsetzen und eine Entwicklungsumgebung, und mich durch die aktuellen Developer-Docs von HA wühlen, etc.., etc.

marcschmiedchen, 2023-02-28:
Schreiben ist implementiert. Freue mich über Feedback, ich konnte natürlich nur testen, was ich selbst hier verfügbar habe, also bspw. keine CWL/BWL etc.

=>Musste stark bei @maxibick spicken, vor allem bei der Implementierung von Config-Flow. Vielen Dank für die Hilfestellung. Hab dich als Co-Autor der Integration mit in das Manifest eingetragen.

closed 2023-02-28 by marcschmiedchen