marcschmiedchen/home-assistant-wolf_ism8 #11: Feedback

lukastripp, 2023-03-03:
@marcschmiedchen  einfach toll, dass es noch implementiert wurde. Noch ein kurzes Feedback von mir.
Läuft alles tip top. Nur muss man erstmal verstehen, wenn man noch die alte Konfig hatte, dass diese raus muss aus der configuration.yaml. Danach klassisch einrichten. 
Bilder habe ich dir auch noch mal angehangen für vielleicht Doku Anpassung noch.

Kann man sich bei dir erkenntlich zeigen mit einer Paypal Spende oder ähnlichen? Ich finde, dass so etwas auch gewürdigt werden muss. Viele fordern immer nur und geben nichts. Kenne das leider viel zu oft. 

Hier noch der Tipp für andere:
Ich habe z.B. eine CGB-2 und das BM-2. 
Wenn man jetzt aber noch die Programme also Werte schreiben möchte, dann muss immer DKW mit ausgewählt werden.

Also für alle HA User, immer DKW & SYM mit auswählen. SYM kann auch manuell einen Datenpush - geniale Funktion.

[image]
[image]
[image]

Dark-Tower-Coder, 2023-03-03:
Von mir auch vielen Dank für die klasse Integration!

marcschmiedchen, 2023-03-09:
Hey leute danke fuer die Screenshots und die tips. Ich übernehm das bei gelegenheit in die doku.
Das mit Paypal is nett gemeint, aber es war/ist ein hobby-projekt und ich hab damit python gelernt (und git). Alles gut.

closed 2023-03-09 by marcschmiedchen