marcschmiedchen/home-assistant-wolf_ism8 #13: Datenpunkte zeigen hin und wieder "None" anstatt reelle Werte

Dark-Tower-Coder, 2023-03-03:
Hallo @marcschmiedchen.

Die neue Version läuft recht gut bei mir.
Eine Sache verstehe ich nicht ganz.

Es gibt ein paar Datenpunkte die bei mir hin und wieder auf "None" gesetzt werden.
Ich weiß aber nicht warum.
Passiert es, weil zu lange keine Daten von der Schnittstelle kommen?

In meinem Screenshot kann man sehen, dass (bis auf ein paar Ausreißer) die Daten korrekt eintrudeln und gespeichert werden.
Aber dann gibt es Momente bei denen der Wert "None" ist.

Da ich die Daten per Automatisierung in Variablen meiner Homematic schreibe, gibt es immer wieder Fehlermeldungen im Log, da es sich hierbei dann um ein String und nicht um eine Zahl handelt.

Wenn das Verhalten korrekt ist, werde ich meine Automatisierungen anpassen müssen.

Vielen Dank und Gruß
Tower

[image]

marcschmiedchen, 2023-03-09:
Hi,
also das wäre ein klassischer Fall fuer debugging, aber ich hab keine CWL mit der ich das testen koennte....

Also rein technisch sollte nie ein "none" zurückkommen, da die Wolf-Library, die mit dem ISM spricht, ein großes Array mit Werten vorhaelt, die immer nur aktualisiert werden wenn eine Message kommt. Die "altern" nicht. Also wenn du das ISM ausstöpselst, sollte die Library ab dann munter immer wieder die gleichen werte zurückliefern, unendlich lange.
Allerdings sind die Werte so lange "none", wenn die Library frisch gestartet wird, bis die erste Message vom ISM kommt (logo). Also bspw. wenn man einen restart des HA-Servers macht oder die Config updatet und HA neu startet.  Das koennte ich unterbinden, in dem ich gleich nach dem Start 1x alle Werte vom ISM8 anfordere; aber wenn HA startet, ist der Rechner eh voll ausgelastet, da wollte cih nicht noch alle Datenpunkte anfordern. Kannst Du ausschließen, dass HA frisch gestartet wurde, wenn diese "None" werte aufauchen?

Dark-Tower-Coder, 2023-03-17:
Hey @marcschmiedchen.
Ich kann mir nicht vorstellen, dass mein HA so of neu gestartet ist, aber wirklich ausschließen kann ich es auch nicht.

Aber ich habe es bei mir jetzt mal beobachtet und seit dem letzten Update konnte ich es nicht wieder sehen.

Also mache ich mal zu.
Vielen Dank :-)

closed 2023-03-17 by Dark-Tower-Coder