marcschmiedchen/home-assistant-wolf_ism8 #15: Datentyp bei Steuerelement "CWL_Programm" ungültig

hwilma, 2023-03-09:
Erstmal vielen Dank für die Umsetzung der Integration. 
Ich hatte vorher eine parallele Installation von FHEM, die dann die Daten via MQTT an Homeassistant übermittelt hat. 
Hat auch funktioniert, war aber alles andere als stabil, geschwige denn komfortabel.

Daher bin ich sehr dankbar für die Umsetzung ;)

Ich habe festgestellt, dass bei der CWL implementierung die Programmauswahl "CWL_Programm" (Auswahl der Lüftungsstufe) auf einen Fehler läuft.

Ich habe die Logs aus Homeassistant angefügt.

[image]

Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/wolf/select.py:151
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 21:34:01 (8 occurrences)
Last logged: 21:49:15

[2762907312] '>' not supported between instances of 'str' and 'int'
[2645677416] '>' not supported between instances of 'str' and 'int'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1845, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 686, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 961, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 726, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 106, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/wolf/select.py", line 151, in async_select_option
    self._ism8.send_dp_value(self.dp_nbr, option)
  File "/usr/local/lib/python3.10/site-packages/wolf_ism8/ism8.py", line 235, in send_dp_value
    if not validate_dp_value(dp_id, value):
  File "/usr/local/lib/python3.10/site-packages/wolf_ism8/ism8_helper_functions.py", line 100, in validate_dp_value
    if (value > max(DP_VALUES_ALLOWED.get(dp_id))) or (value < min(DP_VALUES_ALLOWED.get(dp_id))):
TypeError: '>' not supported between instances of 'str' and 'int'

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: wolf_ism8.ism8_helper_functions
Source: custom_components/wolf/select.py:151
Integration: Wolf Climate Control ISM8 (documentation, issues)
First occurred: 21:34:29 (5 occurrences)
Last logged: 21:45:12

value for 153 has invalid datatype <class 'str'>, should be <class 'int'>
value for 158 has invalid datatype <class 'str'>, should be <class 'int'>

marcschmiedchen, 2023-03-10:
ja , sind gleich 2 Bugs.
Ich behebe es

marcschmiedchen, 2023-03-12:
Habe die zugrundeliegende Library aendern müssen, die neue Version ist bei Pypi hochgeladen (2.0.9). Die Probleme mit den falschen Programmeinträgen sollten gelöst sein, überhaupt hatte die Programmeinstellung noch gar nicht funktioniert. Datenpunkte 153,158 sollten jetzt einstellbar sein aber ich kann es nicht testen, da ich keinen CWL habe. Die neue Version der Integration zieht die neue Bibliothek auch automatisch an/lädt sie runter. Aber für HACS habe ich noch keine tags erstellt, das heißt ihr müsst es erst mal manuell installieren. Der Hintergrund ist, dass ich keine CWL habe und kein richtiges Testen durchführen kann und daher die neue Version erst "offiziell" mache, wenn ich sicher bin, dass beim blind programmieren nichts schief gegangen ist. Freue mich auf Feedback ob die CWL jetzt korrekt angesprochen wird.

hwilma, 2023-03-14:
Habe gestern noch ein bisschen getestet. ;)

Der Parameter CWL_Zeitprogramm lässt sich ohne Fehlermeldungen setzen. Hier scheint das Format zu passen. 
Die Übertragung an die Lüftungsanlage funktioniert trotzdem leiden nicht.
Auch beim System Bedienmodul kann ich keinen Parameter übertragen. 
Gibt es hier detaillierte Logfiles, die man ziehen kann? Im "normalen" log finde ich keine Einträge.
Der Empfang der Daten läuft problemlos.

Darüber hinaus sind mir zwei Themen mit der Formatierung der Daten in der CWL aufgefallen.
Die Lüftungsstufe gibt einen Wert in Prozent zurück, hier ist in der Integration aktuell ein string-Wert hinterlegt.
Der zweite Punkt ist der Luftdurchsatz. Hier wird der wert mit Faktor 10.000 zu groß angezeigt. Darüber hinaus wurde ein unplausibler Wert übertragen. (Siehe Bilder)

[image]

[image]

[image]

marcschmiedchen, 2023-03-14:
Haengt alles mit den (größeren) Änderungen in der neuen Lib zusammen. Ich hoffe ich kriege es heute hin. Beim mir funktioniert schon das Setzen der Zeitprogramme (war ein dummer Bug). Den Faktor 10.000 hab ich auch schon einprogrammiert, mangels CWL aber nicht getestet ob es klappt. Sollte aber. Ich lade demnächst eine neue Version 2.1 für BEIDES hoch (Library und die INtegration), und bin dann nochmal auf Feedback angewiesen. Danke fürs Beta-Testen.

marcschmiedchen, 2023-03-15:
So alle Fehler in diesem Issue sollten behoben sein. Sollte jetzt auch über HACS funktionieren/gezogen werden.
Den Fehler mit den Prozentwerten, die als String implementiert sind habe ich nicht ....verstanden(?)... Die Lib liefert hier float-Werte. Wenn da doch noch was offen ist, bitte einen neuen Issue aufmachen damit ich die Fehler besser separieren kann.
gruss
MS

closed 2023-03-15 by marcschmiedchen