marcschmiedchen/home-assistant-wolf_ism8 #16: Datentyp für prozentuale Leistung (Heizgerät oder CWL Excellent) ohne Einheit

hwilma, 2023-03-16:
Der Datentyp für die prozentuale Leistung des Brenners oder der Lüftungsanlage wird nicht als grafischer Verlauf dargestellt, das liegt möglicherweise an der fehlenden Einheit "%"
Die prozentuale Leistung wird in der aktuellen Version ohne Einheit als Leistungsfaktor interpretiert und in Homeassistant angelegt.

[image]

[image]

In Homeassistant wird die Historie nicht als grafischer Verlauf sondern als Zustandsverlauf dargesetllt (wie bei Text-Sensoren):
[image]

Grundsätzlich wäre die Einheit in Prozent korrekt, ich bin mir aber nicht sicher, ob das die Problematik mit der Anzeige behebt.
Siehe Datenblattauszug ISM8:
[image]

lzier, 2023-03-19:
Ich habe testweise in meiner privaten Installation in sensor.py Folgendes ergänzt:
class WolfScaleSensor(WolfBaseSensor):
    @property
    def unit_of_measurement(self) -> str:
        """Return the unit of measurement of this entity."""
        return "%"

Das Ergebnis sieht gut aus, wobei ich nicht weiß, ob es unerwünschte Seiteneffekte hat. 
[image]

marcschmiedchen, 2023-03-19:
Danke fuer den Hinweis! Als "Unit" muss man "PERCENTAGE" zurückgeben, dann geht es. Ich ergänze es demnächst.

marcschmiedchen, 2023-03-21:
Ist behoben