marcschmiedchen/home-assistant-wolf_ism8 #17: Datentyp für AN/ AUS fehlerhaft

hwilma, 2023-03-16:
Die folgenden Schalter führen beim Betätigen zu einer Fehlermeldung
  - CWL_Zeitw. Feuchteschutz AN_AUS
  - CWL_Intensivlueftung AN_AUS

Logger: homeassistant.components.websocket_api.http.connection
Source: components/select/__init__.py:105
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 15:09:16 (6 occurrences)
Last logged: 22:23:26

[2632173432] Option 1 not valid for select.cwl_zeitw_feuchteschutz_an_aus
[2632173432] Option 0 not valid for select.cwl_zeitw_feuchteschutz_an_aus
[2589123128] Option 1 not valid for select.cwl_intensivlueftung_an_aus
[2647202712] Option 1 not valid for select.cwl_intensivlueftung_an_aus
[2647202712] Option 1 not valid for select.cwl_zeitw_feuchteschutz_an_aus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1845, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 686, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 961, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 726, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 105, in async_select_option
    raise ValueError(f"Option {option} not valid for {entity.entity_id}")
ValueError: Option 1 not valid for select.cwl_intensivlueftung_an_aus

hwilma, 2023-03-16:
Habe in den Logs noch folgende Fehlermeldung gefunden die ziemlich häufig auftritt. Vielleicht steht diese mit dem Problem im Zusammenhang:

Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:137 
First occurred: 08:07:39 (452 occurrences) 
Last logged: 23:14:30

Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 876, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.10/site-packages/wolf_ism8/ism8.py", line 137, in data_received
    self.process_msg(data[_header_ptr + 10 : _header_ptr + msg_length])
  File "/usr/local/lib/python3.10/site-packages/wolf_ism8/ism8.py", line 164, in process_msg
    self.decode_datapoint(dp_id, dp_raw_value)
  File "/usr/local/lib/python3.10/site-packages/wolf_ism8/ism8.py", line 177, in decode_datapoint
    Ism8.log.error("unknown datapoint: %s, data:%s", dp_id, result)
UnboundLocalError: local variable 'result' referenced before assignment

marcschmiedchen, 2023-03-19:
Also die erste Fehlermeldung:  da bin ich noch auf der Suche. Theoretisch sollten die beiden Punkte "0" und "1" akzeptieren. Ich weiss echt nicht wo das Problem ist. Kannst Du bitte mal sicherstellen, dass Du die beiden komplett aus HA gelöscht hattest und dann wieder neu hinzufuegst?
Problem 2: hat nichts damit zu tun, bedeutet aber, dass dein ISM8 Daten sendet, die meine Version der Doku noch nicht kennt/kannte und die ich nicht implementiert habe. Ungluecklicherweise ist im "log"-Befehl ein fehler gewesen, und statt den fehlenden Datenpunkt zu loggen hat er nur den Fehler geworfen. Den Fehler korrigiere ich in der nächsten Version, dann kommt die Log-Meldung korrekt durch und DANN koennen wir den fehlenden Datenpunkt einbinden.

marcschmiedchen, 2023-03-21:
Habe es korrigiert.

hwilma, 2023-03-21:
Hallo marcschmiedchen,

Ich habe die Integration komplett neu hinzugefügt. Beim Setzen der Werte für die betreffenden Datenpunkte bekomme ich nun folgende Fehlermedlung:

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: wolf_ism8.ism8_helper_functions
Source: custom_components/wolf/select.py:137
Integration: Wolf Climate Control ISM8 (documentation, issues)
First occurred: 20:57:29 (3 occurrences)
Last logged: 21:08:54

value for 153 has invalid datatype <class 'str'>, should be <class 'int'>
value for 158 has invalid datatype <class 'str'>, should be <class 'int'>

Vielen Dank für die Bemühungen.

[image]

[image]
[image]