marcschmiedchen/home-assistant-wolf_ism8 #20: unbekannte Datenpunkte erzeugen Fehler im Log.

hwilma, 2023-03-21:
Nach dem Bugfix werden nun die Datenpunkte aufgelistet, die nicht bekannt sind.
Die Punkte sind nicht in der Doku hinterlegt, könnte also auch ein Bug des ISM8 Moduls sein.

Logger: wolf_ism8.ism8
Source: runner.py:179
First occurred: 20:57:03 (144 occurrences)
Last logged: 21:28:15

unknown datapoint: 343, data:bytearray(b'\x00')
unknown datapoint: 344, data:bytearray(b'\x00\x00')
unknown datapoint: 345, data:bytearray(b'\x00\x00')
unknown datapoint: 347, data:bytearray(b'\x00')
unknown datapoint: 348, data:bytearray(b'\x00')

bhuddah, 2023-10-30:
Ich sehe auch unbekannte Datenpunkte:

```
2023-10-30 12:30:40.787 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 336, data:bytearray(b'\x00')
2023-10-30 12:30:40.793 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 362, data:bytearray(b'\x00')
2023-10-30 12:30:40.795 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:bytearray(b'\x00')
2023-10-30 12:30:40.887 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 336, data:bytearray(b'\x00')
2023-10-30 12:30:40.893 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 362, data:bytearray(b'\x00')
2023-10-30 12:30:40.897 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:bytearray(b'\x00')
2023-10-30 12:30:40.989 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 336, data:bytearray(b'\x00')
2023-10-30 12:30:40.995 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 362, data:bytearray(b'\x00')
2023-10-30 12:30:41.000 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:bytearray(b'\x00')
2023-10-30 12:30:41.091 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 336, data:bytearray(b'\x00')
2023-10-30 12:30:41.097 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 362, data:bytearray(b'\x00')
2023-10-30 12:30:41.101 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:bytearray(b'\x00')
2023-10-30 12:30:41.193 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 336, data:bytearray(b'\x00')
2023-10-30 12:30:41.199 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 362, data:bytearray(b'\x00')
2023-10-30 12:30:41.204 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:bytearray(b'\x00')
2023-10-30 12:30:41.295 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 336, data:bytearray(b'\x00')
2023-10-30 12:30:41.301 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 362, data:bytearray(b'\x00')
2023-10-30 12:30:41.305 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 363, data:bytearray(b'\x00')
2023-10-30 12:30:41.510 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 337, data:bytearray(b'\x03')
2023-10-30 12:30:41.611 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 337, data:bytearray(b'\x03')
2023-10-30 12:30:41.675 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 338, data:bytearray(b'\x01')
2023-10-30 12:30:41.678 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 341, data:bytearray(b'\x00')
2023-10-30 12:30:41.712 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 337, data:bytearray(b'\x03')
2023-10-30 12:30:41.775 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 338, data:bytearray(b'\x01')
2023-10-30 12:30:41.780 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 341, data:bytearray(b'\x00')
2023-10-30 12:30:41.802 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 339, data:bytearray(b'\x00')
2023-10-30 12:30:41.813 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 337, data:bytearray(b'\x03')
2023-10-30 12:30:41.876 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 338, data:bytearray(b'\x01')
2023-10-30 12:30:41.881 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 341, data:bytearray(b'\x00')
2023-10-30 12:30:41.903 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 339, data:bytearray(b'\x00')
2023-10-30 12:30:41.914 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 337, data:bytearray(b'\x03')
2023-10-30 12:30:41.966 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 340, data:bytearray(b'\x01')
2023-10-30 12:30:41.977 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 338, data:bytearray(b'\x01')
2023-10-30 12:30:41.984 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 341, data:bytearray(b'\x00')
2023-10-30 12:30:42.005 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 339, data:bytearray(b'\x00')
2023-10-30 12:30:42.016 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 337, data:bytearray(b'\x03')
2023-10-30 12:30:42.067 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 340, data:bytearray(b'\x01')
2023-10-30 12:30:42.078 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 338, data:bytearray(b'\x01')
2023-10-30 12:30:42.085 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 341, data:bytearray(b'\x00')
2023-10-30 12:30:42.106 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 339, data:bytearray(b'\x00')
2023-10-30 12:30:42.111 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 342, data:bytearray(b'\x01')
2023-10-30 12:30:42.113 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 343, data:bytearray(b'\x01')
2023-10-30 12:30:42.169 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 340, data:bytearray(b'\x01')
2023-10-30 12:30:42.180 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 338, data:bytearray(b'\x01')
2023-10-30 12:30:42.187 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 341, data:bytearray(b'\x00')
2023-10-30 12:30:42.208 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 339, data:bytearray(b'\x00')
2023-10-30 12:30:42.211 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 342, data:bytearray(b'\x01')
2023-10-30 12:30:42.214 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 343, data:bytearray(b'\x01')
2023-10-30 12:30:42.262 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 344, data:bytearray(b'\x01^')
2023-10-30 12:30:42.270 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 340, data:bytearray(b'\x01')
[... cut ...]
2023-10-30 12:30:44.769 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 348, data:bytearray(b'\x01')
2023-10-30 12:30:44.869 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 347, data:bytearray(b'\x01')
2023-10-30 12:30:44.870 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 348, data:bytearray(b'\x01')
2023-10-30 12:30:44.970 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 347, data:bytearray(b'\x01')
2023-10-30 12:30:44.971 ERROR (MainThread) [wolf_ism8.ism8] unknown datapoint: 348, data:bytearray(b'\x01')
```

marcschmiedchen, 2023-11-19:
Hab ich jetzt schon von mehreren Leuten gehört, dass mit den Datenpunkten in der Ecke 330-350... keine Ahnung, ist auch von WOlf nicht rauszukriegen...

bhuddah, 2023-11-20:
Ich würde vorschlagen, diese Region von Werten erstmal zu ignorieren und die Log-Messages auf Debug runterzustufen. Hat schon mal jemand mit Wolf Kontakt aufgenommen? Falls nein, würde ich mich freiwillig melden.

marcschmiedchen, 2024-01-18:
In der neuen Version 3.0 gibt es statt Fehlermeldungen die Möglichkeit, als "DEV" device die Daten als Integer durchzureichen und zu verwenden. Bin über Feedback froh wenn jemand rausfindet, was die Daten darstellen.

closed 2024-01-18 by marcschmiedchen