marcschmiedchen/home-assistant-wolf_ism8 #29: BM2: Bedienmodul BM2 - Wurde ausgewählt, aber nicht als Gerät angezeigt.

LHBL2003, 2024-01-06:
Hallo @marcschmiedchen, danke für die Integration. Ich hoffe Deutsch ist ok, da ich die Vermutung habe, dass Schmiedchen aus Deutschland kommt ;)

Ich benutze FHEM und ioBroker und prüfe gerade, ob ich auf HomeAssistant umsteigen kann.
Dabei ist mir natürlich meine Wolf Heizung wichtig. Momentan läuft mein Testaufbau auf einem Raspberry Pi5, den ich mir extra für HomeAssistant gekauft habe. Das Interesse an der Integration ist also groß ;)

----------------------------------------------------------------------------------
**Erfahrungswert: Weitere Geräte erweitern, mit kleiner Hürde**

Die Einbindung hat auf Anhieb funktioniert, auch wenn ich zur Erweiterung der Geräte (SM1, MM1) alle Geräte einmal löschen und neu anlegen muss, sowie die YAML-Konfiguration überprüfen und den HomeAssistant auch dort neu starten muss.
[image]
[image]

----------------------------------------------------------------------------------
 **Problem: BM2 - Bediengerät 2 / Bedienmodul 2 wird nicht aufgelistet**

Mein aktuelles Problem ist, dass ich BM2 ausgewählt habe, es aber nicht aufgelistet wird. Da es das letzte Gerät ist, hoffe ich, dass es ein Fehler im Programm ist und das letzte Array nicht verarbeitet wird. (Vermutung, habe noch nicht in den Quellcode geschaut)
[image]

[image]

Ist dies ein Bug oder muss ich noch etwas beachten?

----------------------------------------------------------------------------------
**Allgemeine Frage / Feature vorschlag: BM2 - Bediengerät 2 / Bedienmodul 2 wird nicht aufgelistet**
 
Und wenn das Bedienmodul vorhanden ist, frage ich mich, ob ich die folgenden Parameter über HomeAssistant steuern kann. Denn unter SYM sehe ich bisher "nur" die Warmwasser Einmalladung.
- Programmwahl Mischer
- Programmwahl Warmwasser
- Sollwertkorrektur
- Einsparfaktor

Die 4 Parameter sind für mich wichtig, da ich derzeit über den ioBrocker meine Heizung bedarfsgerecht steuere. 
z.B.: alle Heizkörper / Fußbodenheizungsventile zu, dann Heizung Standby
oder: Mind. 1 Heizkörper 100% auf, dann mehr Leistung, um den Raum schnell warm zu bekommen.
usw.

[image]

Vielen Dank für deine Hilfe.

LHBL2003, 2024-01-06:
Das was ich vom Bedienmodul gesucht habe, befindet sich scheinbar unter. DKW und dort kann man auch die Sachen via Dropdown auswählen. Somit haben sich meine Punkte oben erledigt.

[image]

closed 2024-01-06 by LHBL2003

JDD-Home, 2024-01-06:
OK ich würde das Fass doch noch mal aufmachen.
Warum kann dich BM2 auswählen, wenn es aber gar nicht als Gerät gelistet wird und scheinbar die Dinge von BM2 in anderen Bereichen wie DKW oder SYM wiederzufinden sind?

marcschmiedchen, 2024-01-08:
Hi,
die Aufteilung richtet sich strikt nach der Wolf-Spezifikation. Muss man googeln, im Moment ist sie auf 
https://www.wolf.eu/de-de/professional/downloads/downloadcenter?selection=ism8
zu finden.

"BM2" kann man auswählen, weil es laut Spec Datenpunkte dazu gibt. Falls die Datenpunkte nicht ankommen, kann das z.B. auch an veralteter Firmware liegen, ich bekomme die Daten bei mir auch nicht... und bei der Sortierung der Dinge habe ich mich streng an die Hersteller-Sortierung gehalten, sonst kommt man nur durcheinander, wenn bspw. jemand von FHEM nach HA umsteigt. So wie es ist, kann man jeden Datenpunkt mit der Wolf-Doku zusammen eindeutig identifizieren....

JDD-Home, 2024-01-08:
Danke für deine Rückmeldung, dann weiß ich ja das alles passt. Vielen Dank für deine mühe. :)