marcschmiedchen/home-assistant-wolf_ism8 #33: no data since 3.0 update

lukastripp, 2024-01-21:
@marcschmiedchen  seit dem update auf 3.0 bekomme ich keine Daten mehr. Habe schon neu eingerichtet und ISM8i Modul neugestartet. Geht nix.  Haben andere auch Probleme?
Mein ism8i Modul:
[image]

nanosonde, 2024-01-21:
Bei mir auch. ISM8i mit CWL. 
Bin wieder zurück auf 2.1.3.

hwilma, 2024-01-21:
Bei mir läuft es sehr gut mit der neuen Version. 
Ich musste jedoch die 0.0.0.0 als IP angeben. Mit der ip vom ISM 8 hat es nicht funktioniert.

Außerdem hat sich die Bezeichnung der Sensoren geändert, wodurch auf der alten Statistik keine Daten mehr erscheinen. Hier müssen einmalig die neuen Namen hinterlegt werden. 
War das möglicherweise das Problem?

lukastripp, 2024-01-21:
Ich habe jetzt mal noch rumprobiert. Ich habe das BM-2 auf Version 3.2 geupdatet. Hat nichts gebracht. Ich habe auch mal 0.0.0.0 oder meine HA IP angegeben bei der Einrichtung ändert nichts. In Version 2.1.3 geht wieder alles. Das die Werte noch mal neu eingerichtet werden müssen, resultiert aus der Änderung, dass Geräte ausgeschrieben sind. Das ist ok. Ist etwas komisch. Wer hat denn welche FW Version auf dem ism8i Modul? @hwilma  wie ist deine, da es ja geht bei dir?

Vainac86, 2024-01-21:
Kann ich auch bestätigen. Selbes Problem.

hwilma, 2024-01-22:
Ich habe folgenden HW/SW Stand:

HW-Version:	 	1.0
FW-Version:	 	1.60

Vielleicht war es auch nur Zufall, dass es irgendwann mit der 0.0.0.0 geklappt hat. 
Ich habe mehrfach die Meldung bekommen, dass ein Fehler in der Konfig vorliegt

lukastripp, 2024-01-22:
@marcschmiedchen  so habe ich das mit der IP auch immer verstanden und gemacht. Ich nutzt die OVA von HA auf einem ESXi. Das mit dem Stromreset vom ism8i ist mir auch bewusst. Ich habe per HACS geupdatet und dann neu eingerichtet,...tot :-(. Habe auch neu start vom HA etc alles gemacht.

nanosonde, 2024-01-22:
Danke @marcschmiedchen für die Erläuterungen.

Ich war wohl zu ungeduldig. 😊

Ich habe mit HACS jetzt nochmal auf HACS aktualisiert. Dann habe ich die Instanz aus HA gelöscht. Nutze das aktuelle HAOS in einer VM.

Dann habe ich als IP 0.0.0.0 gelassen und nur CWL, BM-2 und Systembedienmodul ausgewählt.

Zuletzt habe ich dann 1x den Button „Datenanforderung“ gedrückt.

Jetzt sind alle Entitäten wieder da.

lukastripp, 2024-01-22:
Ich habe das jetzt auch noch mal gemacht. Debug Log ist an. Es passiert nichts. Gibt es noch ein anderes Log? Nur das habe ich im LOG:
2024-01-22 17:01:38.893 DEBUG (MainThread) [custom_components.wolf] Listening for ISM8 on ('0.0.0.0', 12004) : 12004
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Brennerleistung on Heizgeraet1 as DPT_Scaling. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Kesseltemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Sammlertemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Ruecklauftemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Warmwassertemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Aussentemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Anlagendruck on Heizgeraet1 as DPT_Value_Pres. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Aussentemperatur on Systembedienmodul as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Raumtemperatur on Direkter_Heizkreis_WW as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Raumtemperatur on Mischerkreis1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Sammlertemperatur on Kaskadenmodul as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Gesamtmodulationsgrad on Kaskadenmodul as DPT_Scaling. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Vorlauftemperatur Mischer on Kaskadenmodul as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Eingang E1 on Kaskadenmodul as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Eingang E2 on Kaskadenmodul as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Abgastemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Kesselsolltemperatur on Heizgeraet1 as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.058 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Vorlaufsolltemperatur dir. HK on Direkter_Heizkreis_WW as DPT_Value_Temp. Write access: 0
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Programmwahl Heizkreis on Direkter_Heizkreis_WW as DPT_HVACMode. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Programmwahl Warmwasser on Direkter_Heizkreis_WW as DPT_DHWMode. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Heizkreis Zeitprogramm 1 on Direkter_Heizkreis_WW as DPT_Switch. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Warmwasser Zeitprogramm 1 on Direkter_Heizkreis_WW as DPT_Switch. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Programmwahl Mischer on Mischerkreis1 as DPT_HVACMode. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Programmwahl Warmwasser on Mischerkreis1 as DPT_DHWMode. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Mischer Zeitprogramm 1 on Mischerkreis1 as DPT_Switch. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Warmwasser Zeitprogramm 1 on Mischerkreis1 as DPT_Switch. Write access: 1
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Stoerung on Heizgeraet1 as DPT_Switch. Write access: 0
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Status Brenner on Heizgeraet1 as DPT_Switch. Write access: 0
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Status Heizkreispumpe on Heizgeraet1 as DPT_Switch. Write access: 0
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Status Speicherladepumpe on Heizgeraet1 as DPT_Switch. Write access: 0
2024-01-22 17:01:39.059 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Status 3W-Umschaltventil on Heizgeraet1 as DPT_OpenClose. Write access: 0
[... cut ...]
2024-01-22 17:01:39.063 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Kesseltemperaturvorgabe on Heizgeraet1 as DPT_Value_Temp. Write access: 1
2024-01-22 17:01:39.063 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Gesamtmodulationsgradvorgabe on Kaskadenmodul as DPT_Scaling. Write access: 1
2024-01-22 17:01:39.063 DEBUG (MainThread) [custom_components.wolf.wolf_entity] setup wolf entity Sammlertemperaturvorgabe on Kaskadenmodul as DPT_Value_Temp. Write access: 1

marcschmiedchen, 2024-01-22:
Das Log sagt eigentlich nur, dass alles korrekt läuft, aber keine Verbindung zustande kommt (nach dem "Listening for ISM8 on ('0.0.0.0', 12004) : 12004" sollte irgendwann noch kommen, dass er eine connection kriegt. (""Connection from ISM8: ...-....-....-...."

D.h. er lauscht, aber das ISM8 baut keine Verbindung auf. Du hast ganz sicher nichts sonst geändert, Maschine gewechselt , IP adressen oder Port gleichzeitig geändert? Läuft auch sicher nicht noch eine zweite/alte instanz, die sich die connection "schnappt"? 

Es gäbe noch Log-einträge die interessant wären, sowas in der Art "Port already in use" oder ähnlich. Dann blockiert ein noch laufender alter client die IP-Adresse und den port.

Das Einrichten mit HACS habe ich nur getestet (benuzte ich sonst nicht), war auch eher hakelig, erst bin ich in so einem Config-Loop hängen geblieben. Fehlerlösung war (vermutlich, bei mir), dass der Neustart, den HACS vorschlägt, nicht ausgereicht hat. 

Kann nur mal raten (die maximalversion): ism8-interface auf einstellungen checken, dann ISM8 neustart. Wolf-Integration aus HA UND HACS entfernen, neustart (in der VM, nicht über die homepage), integration wieder über HACS hinzufügen, nochmal neustart der VM (oder mindestens HA über die shell). Und dann erst konfigurieren. 
Sorry, aber ich hab leider keine ideen sonst. Kann auch an Python/Asyncio-Versionen liegen... (?) Ich checke nochmal die Library Doku ob das mit 0.0.0.0 auch in neuen versionen noch ok ist....

lukastripp, 2024-01-23:
Per root direkt auf der HA VM:
Port 12004 ist da. Wenn ich das Plugin lösche ist die Session auch zu.
# netstat -an |grep 12004
tcp        0      0 10.200.2.251:12004      0.0.0.0:*               LISTEN

Dann hatte ich noch die Multicast Adressen geprüft:
Das Haupt Interface:
2:      enp3s0
        inet  224.0.1.187
        inet  239.255.255.250 users 5
        inet  224.0.0.251 users 3
        inet  224.0.0.252
        inet  224.0.0.1
# cat /proc/net/igmp
Idx     Device    : Count Querier       Group    Users Timer    Reporter
1       lo        :     2      V3
                                FB0000E0     1 0:00000000               0
                                010000E0     1 0:00000000               0
2       enp3s0    :     5      V2
                                BB0100E0     1 0:00000000               1
                                FAFFFFEF     5 0:00000000               1
                                FB0000E0     3 0:00000000               1
                                FC0000E0     1 0:00000000               1
                                010000E0     1 0:00000000               0

Mir scheint als würde die Multicast Group nicht mehr gehen. Da die Daten, so habe ich das verstanden bei MCast geschickt werden. 

Es hat alles bis zum Upgrade auf v3.0 tadelos funktioniert. Es wurde nichts geändert, ausser eine neue Glasfaserleitung an der Firewall. Hat aber nichts mit "LAN" dann zu tun.

Komisch ist jetzt nur, das auch die aältere Version nicht mehr ans fliegen kommt, nach dem hin und her probieren.

marcschmiedchen, 2024-01-23:
Die Kommunikation ist, soweit ich weiss, kein Multicast, sondern eine ganz normale TCP/IPv4 Verbindung: der ISM8 hat über sein Webinterface die Zielserver IP gespeichert (in Deinem Fall hoffentlich 10.200.2.251) und schickt die Daten ausschließlich dahin. Man darf also mit seiner VM auch nicht einfach die IP wechseln. (DHCP=neinnein)

Kannst du von der VM aus die ISM8 anpingen?
wenn ja -> hardcore: mit wireshark auf der VM lauschen, ob was vom ISM8 ankommt.

Scheint mir primär ein netzwerk-problem, vielleicht im zusammenspiel mit der VM zu sein. Dafür spricht auch, dass die alte Version nicht mehr geht ? Bei allen anderen (nanosonde, hwilma, und die leute mit den anderen problemen :)) läufts. 

Ich will nochmal drauf hinweisen: wenn man per HACS installiert ist keinesfalls der angebotene Neustart von HA ausreichend. Man muss die VM hoch und runterfahren. Oder als root HA neu starten. Keine Ahnung wieso. Ich hatte sonst immer einen config-loop.

lukastripp, 2024-01-26:
Es geht. Das Modul hat scheinbar sich verschluckt. Erst nach IP auf eine andere und wieder zurück mit jeweils Neustart der Heizung hat's gebracht.

closed 2024-01-26 by lukastripp