marcschmiedchen/home-assistant-wolf_ism8 #34: Problems controlling functions since 3.0

moonGbox, 2024-01-23:
Hi,
since I did an update to version 3.0 I have problems to control the features of my CWL from Home Assistant. Before it was working without problems.

Since the update I'm not able to start for instance the "Intensivlüftung".  The optionlist offers "1" to start the "Intensivlüftung" and "0" to turn it off. If I start the "Intensivlüftung" directly on the Wolf-Device, im able to turn it off from Home Assistant, but not to turn it on again. All the same with other optionlists. Did you change anything within the function handling?

thanks and best regards,
moongringo

hwilma, 2024-01-23:
I can confirm this issue. My CWL can not be controlled by the integration since updating to Version 3.0.

marcschmiedchen, 2024-01-23:
Hm, guys we need to clarify a bit....

1) "all the same with other optionlists", even though they are intrinsically implemented very very different?  Are you sure that NONE of them are working. I.e. "Zeitprogramm 1/2/3" or  "Programm" Selects are also not working? This problem should extend to all other Devices, not only CWL, but on HG/BWL/MM/SM also since the implementation is the same. I would hope that only "Intensivlüftung" and "Feuchteschutz" are not working? Their Implementation is unique to the CWL...
2) do I read correct, the "switching off" works, but not "switching on"? seems to be a problem with a bool <-> int conversion. I will look into it.

marcschmiedchen, 2024-01-27:
I found the bug. Happens when int (1) is casted to boolean (true). Will fix it.

marcschmiedchen, 2024-02-06:
fixed it. (cannot test => I'm closing the issue in good faith :)
feel free to re-open it if error persists

closed 2024-02-06 by marcschmiedchen

moonGbox, 2024-02-06:
Hi Marc,The fix works perfectly!Thank you!Von meinem iPhone gesendet