marcschmiedchen/home-assistant-wolf_ism8 #35: not a valid add-on repository

RFA81, 2024-01-29:
Wenn ich das Addon hinzufügen will erhalte ichfolgende Meldung

https://github.com/marcschmiedchen/wolf_ism8 is not a valid add-on repository

RFA81, 2024-02-03:
Hi @ All, 
Addon now available in HACS, but it will not show up in the system

marcschmiedchen, 2024-02-05:
hi,
as you found out already: the integration is now available in the HACS store now, so no need to copy/paste the URL, (in fact, it does not let you install it in this way any longer).

The workflow for installation would be : open HACS-> search & select the wolf integration-> press download -> REBOOT!! /restart. After reboot: head to Settings->devices -> "add integration" -> search for "wolf" (again) -> and install it. 

I didnt know that HACS basically just downloads the files, it does not trigger an installation for you. Maybe that was the problem, could you please check it? Also, it seems to be essential to restart your HA-Server. Sometimes the "restart Home-Assistant" from the GUI does not seem to be sufficient. 

I will update the installation instructions accordingly.

RFA81, 2024-02-06:
Thanks for your checkup. 
I setup HA again and then it works.