marcschmiedchen/home-assistant-wolf_ism8 #36: Programmauswahl Heizkreis, Warmwasser wird gesetzt aber falsch in HA dargestellt.

LHBL2003, 2024-02-03:
Ich kann die Programmauswahl Heizkreis und Warmwasser an die Wolf Heizung übertragen. Aber nach Auswahl wird mir weiterhin der falsche Status in HA angezeigt. Auch nach ein Regenschirm der App oder im Browser steht nicht das was in der Heizung gesetzt ist. 

[image]
[image]

[image]

marcschmiedchen, 2024-02-05:
Hi, das problem kann ich leider nicht replizieren. 
Ich war grade im Keller, hab etwas verstellt und sowohl Heizkreis als auch Warmwasserprogramm updaten sich danach in HA. Allerdings musste ich explizit die Seite im Browser neu laden.
=> bitte nochmal checken mit der neuen Version (3.0.1, die kommt bald)

LHBL2003, 2024-02-05:
Verstell mal bitte etwas in HA. Es wird in der Heizung ankommen, aber nach einem Refresh von der HA Seite wird es in HA nicht mehr passen. Bei mir steht dann spätestens wieder der alte Wert dort, obwohl es in der Heizung angekommen ist.

LHBL2003, 2024-02-05:
Hallo @marcschmiedchen ,

Also ich bin auch nochmal in den Keller gegangen :D 
Ich habe an der Wolf Heizung auf der Seite "Heizkreis" und "Warmwasser" geprüft, ob diese im Automatikbetrieb sind. Warmwasser hatte ich gestern wohl vergessen. Zum Glück haben wir einen Kamin :D
Demzufolge habe ich Warmwasser in den Modus "Automatikbetrieb" geändert.
Home Assistant zeigte nach ca. 1 Minute auch für Warmwasser den Status "Auto" an.

In HA ist nun im Bereich "Direkter_Heizkreis_WW" die "Programmauswahl Warmwasser" und "Programmauswahl Heizkreis" auf "Auto".

Danach habe ich beides in HA auf Standby gesetzt.
Ein paar Sekunden später stand in HA immer noch "Standby".
Nach einem Refresh des Browsers stand in beiden Feldern aber wieder "Auto".
Die Wolf Heizung hat wie erwartet für den Heizkreis und das Warmwasser auf "Standby" umgeschaltet.
In der HA Visu bleibt es jedoch auch nach 2-3 Minuten im falschen Zustand "Auto".

Vielleicht hilft dir das zum Nachstellen des Fehlers. :)

marcschmiedchen, 2024-02-06:
OK, war schwer zu finden. Ein Cache-Problem. Sollte jetzt passen (in 3.0.1)

closed 2024-02-06 by marcschmiedchen

LHBL2003, 2024-02-06:
Funktioniert wie ein Schweißer Uhrwerk Danke schön ;-)