marcschmiedchen/home-assistant-wolf_ism8 #41: CHA Betriebsart

kenodai, 2024-03-26:
Moinsen,
ich hab seite heute meine CHA-07 mit ISM8 am laufen. Leider kriege ich keine Betriebsart angezeigt, sollte laut Doku Datenpunkt 177 sein.

Gruß,
Marc

marcschmiedchen, 2024-03-28:
tatsaechlich ein Bug :) dauert ein bisschen

kenodai, 2024-03-30:
Kein Problem. Gib bescheid, wenn du ne WP zum testen brauchst.

marcschmiedchen, 2024-04-05:
sollte behoben sein, version 3.0.2.
Bitte mal testen und ggf den Issue zu machen

kenodai, 2024-04-17:
Sorry für die späte Antwort.

Er kennt nur zwei Stati. Standby und Heizbetrieb. Desweiteren scheinen seit 3.0.2 die Parameter Heizleistung und Leistungsaufnahme defekt zu sein.

marcschmiedchen, 2024-04-18:
also implementiert ist nach Doku für BWL-1S/CHA auch nur folgendes:
    0: "Schornsteinfeger / LegioProtect",
    1: "Heizbetrieb",
    2: "Vorwärmung",
    3: "Aktive Kühlung",
    6: "Standby",
    7: "Test",
    11: "Frostschutz",

kommt bei dir also wirklich nur 1 und 6 als betriebsart durch? aber wenn die durchkommen gibt es keinen grund, warum andere nicht übersetzt würden. Ich würde <fast> vermuten, dass deine CHA dann eben sonst nix schickt. Vielleicht kann jemand mit einer BWL mal erzählen, ob er andere Betriebsarten sieht?

Was den 2. Punkt angeht: Da hab ich absolut nichts geändert, und Heizleistung bei meiner CGB bspw. kommt weiterhin durch. Kannst Du checken, dass sich die Entität in HA auch nicht umbenannt hat ? einfach mal bei entwicklerwerkzeuge/zustände direkt nach der "heizleistung" suchen, bitte. Wenn es sie gibt: was hat sie für nen wert?

kenodai, 2024-04-19:
Ich guck mir beides mal genauer an... Ich lasse mal tshark nen paar Stunden mitlaufen. Und schaue, was da so an Daten rein kommt.

kenodai, 2024-04-19:
Ok, sehr interessant.
Zur 177: Es kam in den letzten paar Stunden tatsächlich nur die 1, dann hab ich die Doku falsch gelesen :)

178/191 war interessant.
In den ersten ca 2,5h kamen 5 Updates, alle mit Length 0. Halte ich mal für normal. Die WP stand still.
Dann hat sich die Wärmepumpe eingeschaltet, erst Heizung, dann Warmwasser und ist seitdem sehr kommunikativ, alle 60-120 Sekunden eine Nachricht. Aber auch die hatten alle Length 0. Im BM2 sind alle Angaben vorhanden.
Hab das ISM8 neugestartet, jetzt passt die Ausgabe wieder.

Falls interessiert, ich hab mir kurz nen dissector für Wireshark zusammengeklimpert:
https://gist.github.com/kenodai/c60acf0cc08744eff151be05ddec16f4

closed 2024-04-19 by kenodai