marcschmiedchen/home-assistant-wolf_ism8 #42: cannot find in HACS store

gasinkas, 2024-04-04:
Hi, I'm unable to find the HACS integration when I search something with "wolf". Do i have to add anything anywhere?

Core 2024.4.0 (also not visible in 2024.3.0)
Supervisor 2024.03.1
Operating System 12.1
Frontend 20240403.1

Best regards,
Stefan

marcschmiedchen, 2024-04-05:
no, should be right there.
If you updated /reinstalled stuff, maybe wait for a day or so, since the HACS integration needs to scan all the repositories, I guess.
Anybody else having issues with HACS?

gasinkas, 2024-04-05:
Here's my view on the HACS search: 

[image]

gasinkas, 2024-04-06:
Sorry, my bad. Thanks for adding that information :)

closed 2024-04-06 by gasinkas