marcschmiedchen/home-assistant-wolf_ism8 #45: I think there is something wrong with the error status of the entities.

LHBL2003, 2024-04-20:
EN:
From a current perspective, I would say that faults are displayed correctly but do not work properly.
For example, if I close the gas valve and restart the heating, I get the message "Problem" for the heater1 fault.

However, if I now create an automation by using the entity "binary_sensor.heizgeraet1_stoerung" as a trigger. (change to "Problem"), the automation is still not started. 
Even if I create a group helper to summarise all faults, this does not react.

I have built a binary_sensor template helper for printing, which is also of the display type Problem. This triggers the automation and also changes the status in my helper group.

DE:
Aus aktueller Sicht würde ich sagen, dass Störungen zwar richtig angezeigt werden, aber nicht richtig arbeiten.
Wenn ich z.B. das Gasventil schließe und die Heizung neu starte, dann bekomme ich als Meldung für das Heizgerät1 Störung „Problem“.

Wenn ich nun aber eine Automatisierung erstelle, indem ich die Entität „binary_sensor.heizgeraet1_stoerung“ als Trigger verwende. (Wechsel zu „Problem“) So wird dennoch die Automatisierung nicht gestartet. 
Auch wenn ich einen Gruppen Helfer erstelle um alle Störungen zusammenzufassen reagiert dieser nicht.

Ich habe mir für den Druck ein binary_sensor Template Helfer gebaut, der auch vom Anzeigetyp Problem ist. Dieser Triggert die Automatisierung und ändert auch in meiner Helfergruppe den Status.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

marcschmiedchen, 2024-05-21:
Should be fixed, but I cannot test it. Please test the issue with the latest version (3.0.2) and close it if errors are gone.

LHBL2003, 2024-05-27:
Unfortunately, the sun shines too often, so the burner doesn't start even after the heating system is restarted. Instead, the mixer briefly reported an error and also sent a push message to the mobile phone. The icon now also fits. See Mischer Icon in Picture 

So I would say it works :)
Thank you. 

[image]

closed 2024-05-28 by marcschmiedchen