marcschmiedchen/home-assistant-wolf_ism8 #56: Warning for 2024-07-11 16:34:36.760 WARNING (MainThread) [homeassistant.helpers.frame] Detected code migth not work with Assistant 2025.1

NorbertMeyer, 2024-07-11:
Let me first say "Thanks a lot" for providing Software and advice how to make use of ISM8 in our new CHA10.

#It took me a day from scratch to get an idea what is needed overall and to get it running on an Docker Desktop instance, after changing the network configuration from "Auto", which was discovered as eth0, to Docker0 fitting better to container running with network=host.

During that I found the following warning in the logfile: 

2024-07-11 16:34:36.760 WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls async_forward_entry_setup for integration wolf with title: ISM8 and entry_id: 01J2HAB2XJGW129NV7WY8MMG1E, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue. 

Without understanding it (yet), I would like to mention before 2025.1 comes.

Again, Thanks a lot

Best regards

Norbert.Meyer@meyers2web.de

marcschmiedchen, 2024-09-06:
thanks, I had that on my list already, but this issue will serve as a reminder

oaken-source, 2024-09-28:
I've just run into the same issue.

marcschmiedchen, 2024-10-01:
I'm on it

marcschmiedchen, 2024-10-04:
fixed it

closed 2024-10-04 by marcschmiedchen

NorbertMeyer, 2024-10-04:
Thanks a lot. 
 Best regards
Norbert--Gesendet mit der 1&1 Mail App