marcschmiedchen/home-assistant-wolf_ism8 #57: Since August eighteen i missing some data from iSM8

AlexJacu, 2024-08-23:
Seit dem 18 August werden mir die Leistungsaufnahme und die Heizleistung meiner Wärmepumpe nicht mehr im Home assistant angezeigt Die Werte verharren bei Null , anhand der Temperatur die immer noch angezeigt wird sehe ich aber wann die Heizung läuft, nämlich etwa alle 12 Stunden. Aber die Sensoren für die Leistungsaufnahme zeigen nichts mehr an , zuvor haben diese Sensoren ganz korrekt angezeigt.
Alle anderen Sensoren scheinen zu funktionieren

marcschmiedchen, 2024-08-23:
Hi,
vielleicht hast du die Integration neu installiert und die Namen der beiden sensoren haben sich aus irgendeinem Grund geändert? Und du schaust jetzt auf die alten Entitäten? Probier mal unter Einstellungen, Geräte, Wolf auf die Wärmepumpe zu klicken und dort alle Entitäten anzuschauen und DORT mal auf die Leistungsaufnahme zu klicken. Wenn da auch keine Werte stehen, schauen wir uns mit dem Debug-Log an, ob übers Netzwerk überhaupt Werte ankommen.

hat sonst noch jemand Probleme mit Wärmepumpen-Leistung?  Wie immer kann ich mangels eigener Wärmepumpe da schwer was zu sagen... Gruss MS

MonsieurJW, 2024-09-04:
Hallo,
ich habe das selbe Problem.
Habe eine CHA-10 und eine CGB-2 als Hybridsystem laufen.
Schon seit der Installation des ISM8I-Moduls habe ich immer mal wieder festgestellt, dass mit einige Datenpunkte "verloren gegangen" sind. Insbesondere die Heizleistung der CHA-10.
Dies konnte ich aber immer bis zum nächsten Mal mit einem Neustart der Anlage beheben.
Nun kommen bei mir seit dem 28.08. keinerlei Daten mehr an.
Ich konnte auf die Schnelle nicht prüfen, ob ich diesem Zeitraum ein Update o.ä. in HA-System gemacht habe.
Möglich wäre es.
Das letzte Update lief gestern auf Core-Version 2024.8.3 und Operatin-System-Version 13.1
Wenn mir jemand sagt, wie ich herausfinden kann, wann die Updates davor liefen, könnte ich schauen, ob dies zeitlich zusammenhängt.
Im Notfall installiere ich Wolf-Integration nochmal neu, was ich aber gerne irgendwie vermeiden würde, da sie in einigen Automatisierungen eingebunden ist.
Aber wenn das am Ende und dauerhaft hilft, muss es natürlich sein.
Hilfe hierzu wäre toll!
Danke!

MonsieurJW, 2024-09-04:
Ich habe für die Integration auch mal den Debugger aktiviert und dabei folgende Einträge gefunden...
Vielleicht hilft das ja?
Es steht eh drin, dass ich das hier reporten soll ;)
[DEBUG_log_Wolf.txt](https://github.com/user-attachments/files/16869238/DEBUG_log_Wolf.txt)

MonsieurJW, 2024-09-05:
Ich konnte meinen dauerhaften Datenverlust korrigieren.
Am Ende war es lediglich das locker sitzende Netzwerkkabel ... (shame on me!!!)
Also wie bei jeder Fehlersuche, einfach immer mal ganz vorne bei den Basics beginnen und das Unmöglichste mit einbeziehen.
Tolle Integration!
Herzlichen Dank dafür und ich gelobe, das nächste Mal vorher genauer hinzuschauen X-)

marcschmiedchen, 2024-09-06:
Danke für das Log, den Fehler hat ein anderer User achon schon reportet (anderes Issue), und ich hatte es bemerkt. Kleiner Fehler der mit auffällt ist, dass überall "write-access: 0" steht. Aber das ist nur informell. Trotzdem ein Bug.

Zurück zum Thema: konnte jemand anders noch Probleme entdecken wie oben beschrieben? i.e. alles läuft, außer CWL-Leistungsaufnahme?  Da das "ganz normale" Daten sind, würde ich eher vermuten, dass das Problem nicht bei der Integration liegt. Besteht die Issue noch?

closed 2024-09-14 by marcschmiedchen