marcschmiedchen/home-assistant-wolf_ism8 #58: Items gone after update

Sailorsbe, 2024-09-09:
Hello
I've made an update with the new version and I'm missing a lot of stuff.
I first deleted the old configuration
I've a CGS2-14/120 gas condensation boiler.
Before the integration detected 4 devices :HG1, Sym, DKW and DKW
Now only 2 devices HG1 and SYM
Before I could change the heating programm, the SHW programm etc. and now this facility has gone.
I restarted several time the ISM8 and HA but no further detection.
Can you help please

marcschmiedchen, 2024-09-23:
Hi, i can only guess since it is very strange that an update would break some entities while others continue to work....

This seems to be an update from a very very recent version though.  There was a feature request some time ago, that the devices should be named with their clear names. Since a very long time now, device HG1 is called "Heizgeraet1", DKW became "Direkter Heizkreis/Warmwasser" etc etc. I can only guess that HA recognized some of your older entities and made the transisition automatically while for some that did not work.

I would recommend looking up your devices in "Einstellungen->Geräte&Dienste->Wolf" and click on the entities list there. hopefully here you will find the missing stuff.

Second recommendation is turning on the log-file and see whats going on. Its in the same menu, called "debug protokollierung aktivieren".

marcschmiedchen, 2024-09-30:
So I guess this resolved the puzzle. Closing the issue.

closed 2024-09-30 by marcschmiedchen