marcschmiedchen/home-assistant-wolf_ism8 #66: Wärmepumpe Steuerung

Hefti1923, 2024-10-23:
In der Integration gibt es leider keine Möglichkeit etwas an den Einstellungen zu verändern. Außer 1x Warmwasser.

Um eine optimiertere Steuerung in Kombination mit z.B. einer PV-Anlage zu erreichen, wäre es super, wenn man den Betriebsmodus für Heizung bzw. Kühlung über Home Assistent einstellen könnte.

Mit freundlichen Grüßen 
Hefti

Vainac86, 2024-10-24:
Ich regel das über Home Assistant, welcher bei bestimmter PV Leistung oder Akku Stand eine Shelly 1 schaltet welche den PV Relais Kontakt einer CHA07 umschaltet. 
Das erhöht dann automatisch die Soll Temperaturen.

procodix, 2024-11-01:
Kann man für eine CHA10 die Stromaufnahme aktuell und Gesamt auslesen?

marcschmiedchen, 2024-11-02:
Ich kann es nicht sicher sagen, sollte aber. Anbei die Specs vom ISM. Da steht CHA: Datenpunkt 191: Leistungsaufnahme in kW
[ISM8i_09_2022.pdf](https://github.com/user-attachments/files/17606601/ISM8i_09_2022.pdf)

marcschmiedchen, 2024-11-15:
Dann mach ich das hier mal zu ....

closed 2024-11-15 by marcschmiedchen