marcschmiedchen/home-assistant-wolf_ism8 #67: Periodischer Datenverlust, hier: Leistungsaufnahme CHA-10

PpyuWuLRmagJnLiQ, 2024-11-10:
Hallo,

ich nutze die Integration jetzt seit dem September mit unserer neuen CHA-10 samt ISM8 und bin an sich super-happy. Ein sehr effektiver Einblick in die Wärmepumpe, die uns schon zu zahlreichen Verbesserungen im Kleinen geführt hat. Leider haben wir jetzt schon 3 mal bemerkt, dass nach einer längeren Betriebszeit auf einmal einzelne Werte nichtmehr übermittelt werden. 

Konkret fiel uns das immer bei der  Wärmepumpe Leistungaufnahme auf, weil wir diese am häufigsten ansehen. Diese fehlt heute wieder seit 4 Stunden. Aber es betrifft auch weitere Daten. Sinn ergibt das eigentlich nicht, denn zumindest die Leistungsaufnahme hängt ja immer auch mit der Heizleistung (wir kühlen nicht) zusammen, und die Heizleistung wird eben konstant weiter übermittelt.

Bei den letzten Malen ebenso wie heute haben wir den Fehler nicht mit "Daten anfordern" beheben können, das schien mehr oder weniger gar keinen Effekt zu haben. Erst ein Neustart der Wärmepumpe (Betriebsart auf Standby, abwarten, stromlos stellen) hat geholfen. Den habe ich heute extra noch nicht gemacht, damit wir so vielleicht mehr erfahren können. Vielleicht könnten wir uns dafür zuerst einmal die Datenübermittlung gemeinsam ansehen? Welche Daten sollen wir hierfür wie auslesen lassen, damit ihr euch das Problem genauer ansehen könnt?

[image]

Vielen Dank und viele Grüße,
Christian

s-weise, 2024-11-14:
Moin, 
ich habe das selbe Problem mit einer CHA-7 und einer CGB-2-20 im Hybridverbund. Die Daten kommen einige Tage/Wochen ohne Probleme dann fehlen die Daten bzw. es wird nur 0 gesendet. Witzigerweise ist es bei mir unter anderem die Heizleistung (0xb2 bzw. 178) die nicht mehr ordentlich gesendet wird. Ich habe mal die Pakete aufgezeichnet und per Wireshark analysiert. Für mich sieht es so aus als ob die Heizung tatsächlich 0 sendet. Deshalb habe ich mal bei Wolf angefragt ob die helfen können, bis jetzt ohne Antwort.

Keine Werte
[image]

Normal
[image]

marcschmiedchen, 2024-11-15:
Hmmm, Ich lass das Thema noch ein paar Wochen offen, damit ihr euch hier austauschen könnt. Was ich ausschließen kann ist ein Fehler der Integration, denn der Rücksprung auf 0 kann nur passieren, wenn die Heizung auch wirklich "0" sendet, sonst würde der alte/letzte Wert weiter bestehen bleiben...  Das wird ja auch durch das Netzwerkprotokoll (vielen Dank dafür) bestätigt, insofern vielleicht wirklich derselbe Fehler.

Vielleicht hilft ein Firmware Update der CHA und des ISM? Müsste halt der Wolf Kundendienst mal kommen und im Zuge der Wartung gleich alles aktualisieren (achtung, bei mir gingen alle Zeitprogramm etc verloren. Vorher aufschreiben!).

AlexJacu, 2024-12-03:
ich habe das gleiche problem festgestellt, ebenfalls bei der Leistungsaufnahme. Es ist also sehr wahrscheinlich ein Problem der Firmware des ism8i

s-weise, 2024-12-04:
Mein BM-2 hat die Firmware 3.20 und das ism8 1.80. Für das ism8 gibt es wohl eine neue Firmware die aber nur Wolf selber flashen kann. Ich hab leider immer noch keine Antwort vom Support, hab jetzt aber noch mal nachgefragt. Villeicht könne die die das Problem auch haben auch ein Ticket bei Wolf aufmachen dann bekommt das Thema mehr Aufmerksamkeit.

chrhen, 2024-12-08:
Habe das Problem ebenfalls, und zwar mit den Parametern _Kühlleistung_ und _Heizleistung_, die als 0 empfangen werden. Im Bedienmodul sind konsistente Werte abzulesen. _Leistungsaufnahme_ hatte bisher keinen Ausfall.

sprangen, 2025-01-05:
Wir haben hier auch Aussetzer beim Empfang der Leistungsaufnahme einer CHA und auch das ism8 mit Firmware 1.80.

marcschmiedchen, 2025-01-06:
Ich hab mir im Zuge anderer Fehlersuche nochmal ganz genau die Bytes des Netzwerkprotokolls angesehen. 

Also 1) tatsächlich gibt das Datagramm an, dass ein Datenpunkt "Länge 0" gesendet wird, das heißt man kann den schon mal verlässlich unterdrücken, bzw. verwerfen. Dann kommen die falschen 0er nicht mehr durch.
2) Derzeit bestätige ich dem ISM8 den Empfang jedes Datagramms, da es noch nie vorgekommen ist, dass eines fehlerhaft oder unvollständig war. Aber in den Specs steht, wenn ich den korrekten Empfang NICHT bestätige, würde der Datenpunkt bis zu 5x wiederholt. Vielleicht , vielleicht, vielleicht würde die CHA beim wiederholten senden dann korrekte Werte schicken. Ich muss das aber blind reinprogrammieren, da ich keine solchen Fehler bei mir sehe.... ich wäre dann hin und wieder auf feedback angewiesen. Würde jemand von euch sich bereit erklären, ab und zu eine beta-version zu installieren und feedback zu geben? Am besten jemand, der nicht ausschließlich per hacs unterwegs ist, sondern auch ein paar python-befehle abschicken kann?

marcschmiedchen, 2025-01-06:
Dazu noch eine 2. Bitte: es gibt einen anderen offenen Issue (#70), in dem 2 User Probleme mit der Integration haben. Könnt ihr da mal posten, ob das bei euch genauso ist? Ich kann die Fehler dort nicht nachstellen.

s-weise, 2025-01-06:
Das mit der Verbindung ist vielleicht die richtige Idee. Bei mir hatte sich das Problem soweit verschlimmert das die Daten nach ein paar Stunden schon nicht mehr richtig gesendet wurden. Zwischen dem ism8 und meiner HA Installation ist eine Wifi Bridge. Diese habe ich jetzt mal von 5GHz auf 2,4GHz umgstellt zwecks besserer Anbinding.  Seit dem habe ich keinen Ausfall der Daten mehr gehabt. Ich werde berichten ob es so bleibt. Mein HA läuft in Docker ohne hacs, ich packe Addons immer manuell in custom_components, könnte also eine Beta testen wenn das Problem wieder auftritt. #70 scheint für mich ein anderes Problem, allerdings verwende ich noch nicht die neuste Version des Addons.

sprangen, 2025-01-07:
> Ich hab mir im Zuge anderer Fehlersuche nochmal ganz genau die Bytes des Netzwerkprotokolls angesehen.
> 
> Würde jemand von euch sich bereit erklären, ab und zu eine beta-version zu installieren und feedback zu geben? Am besten jemand, der nicht ausschließlich per hacs unterwegs ist, sondern auch ein paar python-befehle abschicken kann?

Erstmal danke für die weiteren Gedanken zur Lösung des Problems.
Ich kann gerne mal eine Version zum Testen installieren und schrecke auch nicht davor zurück ein paar python Befehle auszuführen. 

> Dazu noch eine 2. Bitte: es gibt einen anderen offenen Issue (https://github.com/marcschmiedchen/home-assistant-wolf_ism8/issues/70), in dem 2 User Probleme mit der Integration haben. Könnt ihr da mal posten, ob das bei euch genauso ist? Ich kann die Fehler dort nicht nachstellen.

Die Fehler die in https://github.com/marcschmiedchen/home-assistant-wolf_ism8/issues/70 beschrieben sind, kann ich hier auch nicht nachvollziehen.

marcschmiedchen, 2025-01-15:
Hi,
es gibt im DEV-branch eine neue Version, die auch die neue Library benutzt. Dort wurde die Netzwerkkommunikation verbessert/verändert. Bitte schaut mal ob das die Probleme mit den 0er behebt. Dazu einfach die Dateien aus dem DEV-Branch nach custom_components/wolf kopieren und HA neustarten. Mit HACS kann man es leider nicht testen. Freue mich auf feedback

Luther80, 2025-01-26:
Hallo. 

Gleiches Problem. CHA7 mit ISM8i mit Fw 1.80. Periodisch einfach keine Daten.

[image]

[image]

MbG.

marcschmiedchen, 2025-02-05:
OK, dann kann ich da nix weiter machen.
Die LIB kann nur lesen was gesendet wird.... nach allem was wir gemeinsam untersucht haben kommt halt einfach nichts übers Netzwerk. Ausschließen würde ich: Netzwerkprobleme (andere Werte kommen ja durch), Programmierfehler (wegen wireshark-auszug). Unwahrscheinlich ist außerdem ein Problem mit dem ISM8 und/oder dessen Firmware (weil das Problem explizit nur bei CHA auftritt). Ich denke daher, die CHA oder deren FW ist das Problem... ist aber nur geraten.

Ich lass den Issue mal auf falls es neue Infos gibt.

chrhen, 2025-02-05:
Ich habe mal etwas breiter recherchiert und bin im [KNX-User-Forum](https://knx-user-forum.de/forum/%C3%B6ffentlicher-bereich/knx-eib-forum/1508916-wolf-heizung-und-l%C3%BCftung-%C3%BCber-knx-steuern/page13) gelandet (wahrscheinlich selber Thread aus dem aus @s-weise seine Infos hat.)

Zusammenfassung der Diskussion:
* Es scheint mit FW 1.80 des ISM8 auch dort Datenhänger zu geben
* Ab FW 1.81 scheinen diese behoben zu sein
* Aktuell ist FW 1.90
* Unschön ist nur, dass man die FW nicht selbst aktualisieren kann, sondern die Platine zu Wolf schicken muss, die das dann für einen erledigen und das Gerät wieder zurücksenden. Eine Mailadresse dazu plus konkreter Kontaktperson ist im Thread angegeben.

Ich sehe es auch so, dass diese Integration nichts weiter beitragen kann, und sehe das Problem eher bei der FW 1.80. Hat inzwischen jemand bereits auf eine neuere Firmware aktualisiert und kann berichten?

s-weise, 2025-02-14:
Moin,
nach langer Zeit ohne Probleme hatte ich jetzt wieder einen Ausfall der Daten. Ich wollte die Version aus den DEV-branch aufspielen, find sie aber nicht. Kann jmd den Link posten, vielen Dank!

Luther80, 2025-02-17:
[image]

Komischerweise läufts jetzt schon längere Zeit gut, oben ist der Wert der die iSM8i ausgibt, unten der vorgeschaltete Shelly 3EM.

Sieht mir netzt nicht so aus als ob da irgendwie Daten fehlen würden. Am 8ten hatte ich komischerweise noch eine Spitze von 167kWh am sensor der ISM. Seitdem habe ich den allerdings aus allen Visualisierungen herausgenommen und scheint jetzt seit dem 9ten ohne Aussetzer zu laufen.

Ich habe trotzdem mal Wolf angeschrieben, die haben mir angeboten die FW zu flashen. Ich muss bei Gelegenheit die Karte mal einschicken.

MbG

marcschmiedchen, 2025-02-28:
Der DEV branch existiert nicht mehr, mittlerweile ist die aktuellste Version regulär produktiv. Wie ich verstehe (auch aus dem KNX-Forum) , ist immer die Kombination CHA und FW1.8 betroffen.  Hab ich zumindest den Eindruck. Wäre nett zu wissen, ob das FW-Upgrade hilft. Wenn ja kann ich den issue dann hier schließen. Bis zum Feedback bleibt er mal offen.

s-weise, 2025-03-05:
> Der DEV branch existiert nicht mehr, mittlerweile ist die aktuellste Version regulär produktiv. Wie ich verstehe (auch aus dem KNX-Forum) , ist immer die Kombination CHA und FW1.8 betroffen. Hab ich [...]

Ich habe jetzt die aktuelle Version getestet, leider habe ich wieder das gleiche Problem. Ich habe jetzt mal bei Wolf angefragt um die Firmware aktualisieren zu lassen. Leider scheint der Wolf Support sehr schlecht erreichbar zu sein. Zu meiner letzten Anfrage habe ich nie eine Antwort erhalten, mal sehen ob es jetzt besser läuft.

Fitzbert-Fitz, 2025-04-12:
Hallo zusammen,
ich möchte anmerken, dass ich auch mit dem WolfLink/ISM7 und WolfSmart HA Integration Datenaussetzer unserer CHA7 sehe. Evtl. liegt das Problem noch tiefer in der Anlage und die CHA sendet falsch/keine Daten auf dem eBus. Ich bin gerade auf ISM8 gewechselt und werde das mal beobachten.

s-weise, 2025-05-12:
Hallo,
kurzes Update, ich habe meinen Controller bei Wolf eingeschickt und jetzt die Firmware 1.90 drauf. Ich werde es jetzt einige Wochen laufen lassen und mich nochmal melden ob das Problem durch die neue Firmware beseitigt wurde.

marcschmiedchen, 2025-12-03:
Hi, also bin froh dass es keine Aussetzer mehr gibt. Dann schliesse ich das Issue. 

Wenn mir @AlexJacu noch Hinweise geben möchte/könnte, was an der CHA-Implementierung noch verbessert werden soll: Bitte ein neues Issue aufmachen. (Er hatte oben erwähnt "allerdings vermisse ich noch immer die steuerungsmöglichkeit der CHA via ism8. im iobroker funktioniert das sehr gut, wie ich gelesen habe.")

closed 2025-12-03 by marcschmiedchen