marcschmiedchen/home-assistant-wolf_ism8 #68: missing statistics for all senors

AlexJacu, 2024-12-03:
no one of the sensors have a statistic for use in statistic diagram.
in the list of statistic sensors in ha no ism8i senors are listed

AlexJacu, 2024-12-03:
der Sensor hat die Daten, aber der Rekorder für die Sensoren ist offenbar nicht aktiviert. Hab ich da irgendwo einen fehler gemacht?
[image]
[image]
[image]

marcschmiedchen, 2024-12-09:
nein, der fehler liegt tatsächlich in der integration. Da fehlen Infos, ohne die die Statistik nicht aktiviert wird. Bin dran.

AlexJacu, 2024-12-23:
> nein, der fehler liegt tatsächlich in der integration. Da fehlen Infos, ohne die die Statistik nicht aktiviert wird. Bin dran.

Das wäre echt klasse. Aktuell zeigt HA für die Entitäten nur die letzten 10 Tage an und alles ältere ist verloren.

marcschmiedchen, 2024-12-29:
fixed that with 4.0 per today

closed 2024-12-29 by marcschmiedchen