marcschmiedchen/home-assistant-wolf_ism8 #70: Nach update Fehler

AlexJacu, 2025-01-02:
Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 01:19:44 (7 Vorkommnisse)
Zuletzt protokolliert: 01:19:44

Error setting up entry ISM8 for binary_sensor
Error setting up entry ISM8 for button
Error setting up entry ISM8 for number
Error setting up entry ISM8 for date
Error setting up entry ISM8 for time
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 133, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 186, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry ISM8 (01J47B532M88QS94E1F28ZMGP4) for wolf.sensor has already been setup!

marcschmiedchen, 2025-01-03:
Den Fehler kann ich leider nicht nachstellen/nachvollziehen.  Sieht aus als wenn die Integration 2x installiert wurde. Gibt es eine definierte Schrittfolge, mit der ich den Fehler provozieren kann? Oder passiert das beim Start von HA? Liegt im Verzeichnis /custom_components die Integration vielleicht versehentlich 2x, in veschiedenen Verzeichnissen?
Mein erster Rat wäre: Integration löschen, HA neu starten, Integration wieder installieren.

1) hat noch jemand das Problem? Oder hat es sich vielleicht schon erledigt? 
2) Wenn de/reinstallieren nicht hilft, bitte nochmal die komplette Log-Datei und/oder eine genaue Beschreibung, wann der Fehler auftritt. Wie wurde es installiert? von Hand oder per HACS?

chrhen, 2025-01-03:
Habe denselben Fehler. Installiert über HACS, die Installation liegt nur einmal im custom_components-Verzeichnis.

Zusätzlich bekomme ich sehr viele Fehler bzgl. "unknown datapoints" und zwar auf 767, 362 und 363.

Beide Fehler treten ab einem HA-Neustart auf und wiederholen sich. Habe eine [Logdatei](https://github.com/user-attachments/files/18303696/home-assistant.log.1.txt) angehängt.

chrhen, 2025-01-04:
Ich habe noch zwei weitere wiederkehrende Fehler in der obigen Logdatei ausfindig gemacht:

`2025-01-02 08:35:50.638 ERROR (MainThread) [wolf_ism8.ism8] decoding of dp 149 data, type DPT_HVACMode_CWL failed` 
`2025-01-02 12:01:51.903 ERROR (MainThread) [wolf_ism8.ism8] decoding of dp 70 data, type DPT_HVACMode failed`

marcschmiedchen, 2025-01-05:
sehr sehr seltsam, ich kann nichts davon nachstellen. (bis aus die fehler mit den "unknown datapoints", die gab es schon immer, ich logge sie nur seit dieser version als fehler. Ich überlege mir etwas, dass sie das log nicht zumüllen). Welche HA version ist es denn, und in welcher Form (also docker, core, etc...) benutzt ihr es?  Was könnte HA dazu bringen, nach 48h nochmal das Setup anzuwerfen? Wird der Server regelmäßig neu gestartet?

chrhen, 2025-01-05:
Danke für die Erklärung mit den unknown datapoints. Man könnte die zunächst mal als Warning loggen und nicht als Error, da es ja auch nicht wirklich ein Fehler ist. Zusätzlich könnte man es auch nur einmal pro Integration-Start loggen.

Der HA Server wird nicht neu gestartet sondern läuft durch. Ich nutze HA in einer VM, aktuell sieht es so aus:

> Core 2024.11.3
Supervisor 2024.11.4
Operating System 13.2

Alle drei sind nicht brandaktuell, ich habe aber die Tage vor, zu aktualisieren und würde dann noch einmal checken.

marcschmiedchen, 2025-01-05:
Das mit den Warnings mache ich so, ja. Das andere: Ich hab auf 2024.8 entwickelt, also wenn, dann muss ich das updaten. :(

marcschmiedchen, 2025-01-05:
Habe es ge-updatet, weiterhin keine Fehlermeldungen bei mir. 
Lese ich das LOG richtig, dass bei euch kein einziger Wert als Update durchkommt und auch keine connection zwischen HA und dem ISM zustande kommt? (die log-einträge sind aus der ISM-Library nicht von HA).
Könntest Du bitte in deine Config.yaml folgende einträge übernehmen und dann nochmal loggen:

logger:
 default: info
 logs:
    custom_components.wolf: debug
    wolf_ism8.ism8: debug

Im moment bin ich etwas ratlos.

chrhen, 2025-01-05:
Ist drin. Kann allerdings dauern, bis der Fehler nach dem Neustart wieder auftritt. Ich melde mich dann.

marcschmiedchen, 2025-01-05:
noch 2 fragen:
1) hat der HA server mehr als eine IP-adresse?

2) Steht unter "Einstellungen"->"Geräte"->"Wolf" in der GUI in leicht rot unterlegt sowas wie: 

Benötigt Aufmerksamkeit
ISM8
Keine Geräte oder Entitäten
Einrichtungsfehler:
Überprüfe die Protokolle

aber untendrunter das ISM8 noch ein zweites mal in weiss/OK?

chrhen, 2025-01-05:
Zu 1: Nein, der Server hat nur eine AP-Adresse.

Zu 2: Bisher dachte ich es sei nicht erwähnenswert weil evtl. selbstverschuldet beim Rumspielen. Vor dem kürzlichen Restart war die Integration tatsächlich rot mit einem weißen Teil drunter. Folgender Fehler wurde geloggt als ich reloaden wollte (was nichts gebracht hat):

> 2025-01-05 21:21:22.131 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ISM8 for wolf
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/wolf/__init__.py", line 43, in async_setup_entry
    await _task
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1572, in create_server
    raise OSError(err.errno, msg) from None
OSError: [Errno 98] error while attempting to bind on address ('0.0.0.0', 14002): address in use

Dennoch kamen Daten weiter aktuell an.

0.0.0.0 ist auch die von mir gewählte Einstellung, wie von dir empfohlen.

virusmoere, 2025-01-12:
Ich habe ein ähnliches Problem seit dem 4.0 Update. Anpassen des Ports hat leider nichts gebracht.

``
Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 13:44:30 (1 Vorkommnisse)
Zuletzt protokolliert: 13:44:30

Error setting up entry ISM8 for wolf
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/wolf/__init__.py", line 43, in async_setup_entry
    await _task
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 1615, in create_server
    raise OSError(err.errno, msg) from None
OSError: [Errno 98] error while attempting to bind on address ('0.0.0.0', 12005): [errno 98] address in use
``

marcschmiedchen, 2025-01-13:
Hi, das ist ein ganz anderes Problem: hier ist beim Wechsel der Integration der alte HA nicht sauber heruntergefahren und der Port bleibt weiterhin blockiert ("address in use"). Das kann passieren wenn der Neustart nicht glatt läuft. Da hilft dann leider nur noch (dann aber sicher) ein Neustart der Maschine. Also nicht nur HA, sondern das Linux neustarten.

marcschmiedchen, 2025-01-13:
Der ursprünglich als Issue eingetragene Fehler war "ValueError: Config entry ISM8 (01J47B532M88QS94E1F28ZMGP4) for wolf.sensor has already been setup!". Das konnte ich bisher nicht nachstellen, auch andere nicht.... ist das immer noch aktuell?

virusmoere, 2025-01-13:
> Hi, das ist ein ganz anderes Problem: hier ist beim Wechsel der Integration der alte HA nicht sauber heruntergefahren und der Port bleibt weiterhin blockiert ("address in use"). Das kann passieren [...]

Hi,

leider tritt das auch nach komplettem Neustart des Hosts weiterhin auf. Eine Änderung des Ports funktioniert bis zum Neustart von HomeAssistant und lässt sich danach auch nach einem Neustart des Hosts nicht nutzen.

Mache ich ein Rollback auf 3.x funktioniert alles (auch nach dem Neustart). Soll ich hierfür ein neues issue öffnen?

marcschmiedchen, 2025-01-13:
ja bitte

marcschmiedchen, 2025-02-05:
@AlexJacu : Kann ich das hier zu machen? Habe einiges an der Robustheit des config-entry-handlings verbessert. Ich konnte zwar den Fehler nicht nachstellen, hab aber abgefangen dass es sich doppelt konfigurieren kann.... Dementsprechend sollte das Problem eigentlich nicht mehr auftreten?

closed 2025-02-28 by marcschmiedchen