marcschmiedchen/home-assistant-wolf_ism8 #71: OSError: error while attempting to bind on address

virusmoere, 2025-01-16:
Hi,

seit dem 4.0 Update habe ich Probleme mit der Erweiterung. Scheinbar kann Version 4.0 keine Sockets mehr öffnen. Mache ich ein Rollback auf 3.x funktioniert alles (auch nach dem Neustart).

Leider tritt das auch nach komplettem Neustart des Hosts weiterhin auf. Eine Änderung des Ports funktioniert bis zum Neustart von HomeAssistant und lässt sich danach auch nach einem Neustart des Hosts/der VM nicht nutzen.

Auszug aus dem Log:

```
Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 13:44:30 (1 Vorkommnisse)
Zuletzt protokolliert: 13:44:30

Error setting up entry ISM8 for wolf
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/wolf/__init__.py", line 43, in async_setup_entry
    await _task
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 1615, in create_server
    raise OSError(err.errno, msg) from None
OSError: [Errno 98] error while attempting to bind on address ('0.0.0.0', 12005): [errno 98] address in use
```

marcschmiedchen, 2025-01-17:
Hi, es gibt eine neue beta-Version im DEV-Branch. Wenn Du die von Hand installieren kannst, könntest du checken, ob dein Problem weg ist.  (Dazu die dateien in den Ordner "custom_components/wolf" kopieren). Sie benutzt eine neue/bessere Version der Library und ist auch robuster was das Netzwerk angeht. Ich konnte zwar deinen Fehler nicht nachstellen, aber es gibt eine gute Chance, dass es damit weg ist.

Oder du wartest bis die anderen Tester Feedback geben und ich den Release regulär/produktiv mache. Wie du möchtest.

marcschmiedchen, 2025-02-05:
Kann ich das hier zu machen? Die 4.1 ist schon länger online.... das sollte behoben sein.

closed 2025-02-28 by marcschmiedchen