marcschmiedchen/home-assistant-wolf_ism8 #72: Possibility to set values?

procodix, 2025-01-19:
Hi, fantastic work so far. Would it be possible to set values, for example warm water temperature via a slider?

Like here: https://www.heizungsforum.de/threads/ism7i-linkhome-oder-linkpro-fuer-cgb-20.10994/#lg=attachment40905&slide=0

marcschmiedchen, 2025-01-20:
short answer to question : yes.
long answer to question: The link is for an ISM7.... ISM8 has different (and less) possibilities. but still... yes.

Are you sure that was the question you had in mind?

procodix, 2025-01-20:
Yes, exactly. 

I thought ISM7 an HomeAssistant can only work through Wolf Cloud and scraping their website but was surprised to learn, that someone figured out a way to communicate with it locally.

closed 2025-01-20 by marcschmiedchen

procodix, 2025-01-20:
I am asking for ISM8. ?!?!

marcschmiedchen, 2025-01-20:
Yes you can set values via sliders. I added a screenshot in the Readme on the first page. However, be advised that an ISM7 has way, way, more datapoints available. As a rule of thumb:
-  with an ISM8 you can set anything which could be accessed in "normal" mode on the BM2. This includes water temperature, mode of operation, certain adjustment factors, and operational modes plus some more gimmicks. You can browse through the file "ism8_constants" for a full list.
-  with an ISM7 you can set every(?) value of the so called "Fachmann Ebene", this is not possible with ISM8. 

But everything which IS available with an ISM8 can be set in Home Assistant via GUI, including sliders.

marcschmiedchen, 2025-01-20:
by the way: now that I learned that someone brought the ISM7 to local operation, I would have bought an ISM7 instead of an ISM8 for myself.