marcschmiedchen/home-assistant-wolf_ism8 #75: Keine Geräte Gefunden

HaagfloSIL8, 2025-02-13:
Hallo Zusammen,

ich bin ziemlich neu in der ganzen Thematik, versuche aber gerade meine Wolf BWL1 zusammen mit der CWL Wohnraumlüftung in meinen Homeassistant einzubinden.

Ich habe das ISM8 am BM2 angeschlossen und per Lan Verbunden auf der Konfigurationsseite des ISM 8 habe ich dann die IP adresse der HO Installation eingegeben. 

bei der Installation in HO habe ich dann die 0.0.0.0 eingegeben und den port bei12004 belassen.

die Installation findet nun auch Geräte, allerdings nur die Lüftung und das ISM8 selbst. dabei werden auch nur 2 Datenpunkte (Filter zurücksetzen und Datenabruf) bereitgestellt.

Hatte jemand schon ein ähnliches Problem und kann mir weiterhelfen?

Danke schon vorab!

marcschmiedchen, 2025-02-19:
Hm, 
also die Integration findet ja nichts von selbst, man muss ihr per Häkchen vorgeben, welche Geräte man hat/sehen will. Also in deinem Fall am besten nochmal löschen, neu hinzufügen und im Schritt "WOLF Komponenten auswählen" alles anhaken was Du hast. Ganz wichtig ist ürbigens, die heizung neu zu starten nachdem man im ISM8 was eingetragen hat. Ist leider so.

Wenn man in HA aufOK klickt, versucht die Integration 30s eine Verbindung aufzubauen. Aber selbst wenn das nicht klappt legt sie eigentlich alle gewählten Datenpunkte an. Das warten dient nur dazu, im Falle einer Verbindung ein paar Sachen vom ISM auszulesen, z.B. die Firmware-Version. Aber es sollte auch ohne gehen....

Wenn das nicht klappt:
Ich will nicht ausschließen dass ein Bug z.B. mit einer neuen Firmware-version oder so nicht zurecht kommt. Kannst du mal bitte auf der Geräte-Seite (unter dem Wolf-Logo) das "Debug Protokoll" aktivieren und dann einmal das ISM8 löschen und gleich wieder installieren und in die Logs gucken? Und vielleicht einen Auszug posten?

marcschmiedchen, 2025-02-28:
Hat das geklappt? Dann schließe ich die Issue?

HaagfloSIL8, 2025-02-28:
Hi Marc,

Das gute vorab- es hat geklappt.

Ich kann dir leider nicht sagen an was es gelegen hat habe mehrfach alles
neu installiert, aber erst als ich dann die IP von 0.0.0.0 auf die richtige
IP der HO Installation gesetzt habe hat es funktioniert.

Danke und Gruß
Flo

Mit freundlichen Grüßen

Florian Haag
_________________________________________________________________________________________

Schauinslandstraße 8
71131 Jettingen
0174/1435121

HaagfloSIL8, 2025-02-28:
> OK danke für den Hinweis dann ergänze ich die Anleitung, dass 0.0.0.0
> nicht in jedem Fall funktioniert !

das wäre hilfreich- ich hab im Ism 8 und in der Integration die IP der HO
Installation hinterlegt

Mit freundlichen Grüßen

Florian Haag
_________________________________________________________________________________________

Schauinslandstraße 8
71131 Jettingen
0174/1435121